w3c-cg/web-nfc #222: A simultaneous tap happened on multiple NFC adapters

leonhsl, 2019-05-06:
About the note following step 9.12.4 of https://w3c.github.io/web-nfc/#the-push-method

> NOTE
Multiple adapters should be used sequentially by users. There is very small likelihood that a simultaneous tap will happen on two or multiple different and connected NFC adapters. If it happens, the user will likely need to repeat the taps until success, preferably one device at a time. The error here gives an indication that the operation needs to be repeated. Otherwise the user may think the operation succeeded on all connected NFC adapters.

Would you help me understand this more? Thanks.

We got a device coming within the range, the device already passed verification, then we initiated a data transfer to it,  but if it happens to be a simultaneous tap, i.e. happening also on other NFC adapters, the transfer will fail? It's due to some error raised by the underlying platform (Android)?

Then we reject push() with "NetworkError"? This error seems not that clear that the user needs to retry and should tap on only one device at a time. Just because we cannot get a more specific transfer error indicating a simultaneous tap from the underlying platform?

kenchris, 2019-05-23:
@zolkis ?

zolkis, 2019-05-23:
Currently it is not specified how multiple adapters should be handled (there was some quite heated discussion about it a few years back, but a simple approach was taken for the sake of simplicity). 
The API is not concerned with adapters. In most cases there is a single [default] adapter (managed by the implementation). In case of multiple adapters present (in addition, if multiple adapters are close enough to trigger NFC operation), the implementations should follow what the underlying platform is doing in such cases (policy comes from there). Just be transparent to the application/user and don't try to be too smart by adapting/changing policies etc. At least not now :).

> Then we reject push() with "NetworkError"?

If the underlying platform fails the operation, yes, the application needs to prompt the user to tap again, so it needs to be told by the implementation there was an error.

leonhsl, 2019-05-23:
I see, our API implementations do not have to be aware of whether it's a simultaneous tap, we just forward the current operation result to the user. Thanks!

closed 2019-05-27 by leonhsl