w3c-cg/web-nfc #224: Should we ignore NDEFMessage.serialNumber for NFCWriter::push()?

DonnaWuDongxia, 2019-05-20:
An attribute |serialNumber| has been added to NDEFMessage dictionary, and it seems only meaningful to NFCReadingEvent.
According to the [the push method in Spec](https://w3c.github.io/web-nfc/#the-push-method),  |serialNumber| in NDEFMessage passed to NFCWriter::push() will be ignored. Should we make a clear statement about this?

kenchris, 2019-05-23:
Yes, you cannot write the serialnumber.

kenchris, 2019-05-23:
```
dictionary NDEFMessage {
  DOMString? serialNumber;
  sequence<NDEFRecord> records;
  USVString url;
};
```

NDEFMessage should be renamed NFCMessage. Maybe it is possible to do

```
dictionary NDEFMessage {
  sequence<NDEFRecord> records;
  USVString url;
};

dictionary NFCMessage : NDEFMessage {
  DOMString? serialNumber;
}
```

DonnaWuDongxia, 2019-05-24:
Two types of message may make it confusing for developers. As the |serialNumber| is only meaningful to NFCReadingEvent, how about move the this attribute to this event, making the NDEFMessage more clean.
interface NFCReadingEvent : Event {
  DOMString serialNumber;
  readonly attribute NDEFMessage message;
};

dictionary NDEFMessage {
  sequence<NDEFRecord> records;
  USVString url;
};

I think |serialNumber|  maybe don't need to be nullable because string can be empty.
Is it definite to rename NDEFMessage to NFCMessage?

zolkis, 2019-05-24:
The UID is not part of NDEF messages, so I would support moving out the tag specific (low-level) information from NDEF message structure.
We could indeed put it to the read event - however, I wonder whether should we generalize this and define a separate structure/interface for low level tag/adapter info, UID (serial number) being one property in that.

Anyway, at the moment the above proposal looks good to me.

commit bd57c264 (2019-06-12): Move serialNumber from NDEFMessage to NFCReadingEvent, fixes #224
commit d6be4e07 (2019-06-12): Merge pull request #227 from DonnaWuDongxia/serialNumber

Move serialNumber from NDEFMessage to NFCReadingEvent, fixes #224
closed 2019-06-12 by zolkis