w3c-cg/web-nfc #228: Update NDEFRecordData typedef to use "double" rather than "unrestricted double".

DonnaWuDongxia, 2019-05-29:
typedef (DOMString or double or ArrayBuffer or Dictionary) NDEFRecordData;

As Reilly has pointed out in the code review[1] that: According to the specification[1] a Number should be expressed as a JSON type record. Also, since JSON cannot represent Infinity and -Infinity as much as it cannot represent NaN[3] the typedef should be updated to use "double" rather than "unrestricted double".

[1]: https://chromium-review.googlesource.com/c/chromium/src/+/1628587/6/third_party/blink/renderer/modules/nfc/nfc.cc#104
[2]: https://w3c.github.io/web-nfc/#data-mapping
[3]: NOTE-4 in 24.5.2 JSON.stringify section in http://www.ecma-international.org/publications/files/ECMA-ST/Ecma-262.pdf

kenchris, 2019-08-05:
To be honest, I am not sure how big a use-case it is writing just numbers and as you can surely do that manually and store it as you want.

I believe we have the support because it is valid in/as JSON- which then means that it makes sense to handle as JSON would handle it

[image]

kenchris, 2019-08-06:
@reillyeon?

reillyeon, 2019-08-06:
I agree. Numbers should be handled as part of JSON.

commit abf8f2dd (2019-08-08): Merge pull request #285 from kenchris/gh-pages

Fixes #228
commit cd34c4bc (2019-08-08): Fixes #228
closed 2019-08-08 by zolkis