w3c-cg/web-nfc #230: web-nfc on IOS

OlivierGrenoble, 2019-06-06:
In IOS13, Apple has opened the possibility to write NFC tags and also to call proprietary NFC commands (For example, for TypeV, see the commands 'writeSingleBlock' and 'customCommand' on https://developer.apple.com/documentation/corenfc/nfciso15693tag).

This could allow to have a web app working on both IOS and Android and reading or writing NFC tags.

Is there some plans to get 'web-nfc' on IOS Chrome version?

Regards

kenchris, 2019-08-02:
That is a question for the Chrome iOS team and not the spec, so closing here. Chrome on iOS uses WebKit and so far Google has not implemented other features on top like Web Bluetooth, so I don't believe this will implement this, sorry

closed 2019-08-02 by kenchris

Munoon, 2020-06-30:
The other day, Apple published a list of functions that they will not implement. One of them is Web-NFC. Unfortunately...
More info [here](https://webkit.org/tracking-prevention/).

zolkis, 2020-06-30:
It is a good thing they care about privacy. The mindless tracking and data collection should end.

However, it would be even better to create an environment for the affected functionality that would make it safer to use and/or raise user alertness in order to mitigate the threats. Just cutting the functionality is hopefully only an intermittent "fix".
There have been attempts to create this kind of environment (e.g. in the SysApps WG), but it's not trivial and in the end browsers draw the line.

Munoon, 2020-06-30:
@zolkis actually, I totally agree with you. In my opinion, web already not really safe and browsers really should do something with it. Unfortunately, most popular browsers like Google Chrome don't really care about it and even make their browser dangerous by adding they own analytics without asking users.
By the way, Safari in new MacOS Big Sur started blocking analytics services and it works great! I am truly proud of Apple for their excellent ability to combine good design and high security.

anssiko, 2020-06-30:
This is a periodic friendly reminder that this repo is for technical discussion on specification development. Comments and requests to browser vendors regarding their products and/or product plans should be directed to their own channels. See e.g. https://web.dev/how-to-file-a-good-bug/ on how to constructively contribute this feedback toward implementations.