w3c-cg/web-nfc #237: Distinguish between NFC not available and not enabled.

riju, 2019-06-27:
This came up while [reviewing](https://chromium-review.googlesource.com/c/chromium/src/+/1677736/5/services/device/nfc/android/java/src/org/chromium/device/nfc/NfcImpl.java#381). 

Present :
The Spec grouped all HW unavailable cases to NOT_READABLE, and data_not_support to NOT_SUPPORTED error, see item-5 and item-6 here https://w3c.github.io/web-nfc/#the-push-method.

Proposal in Android parlance, 

1. No NFC chip on device --> (mNfcManager == null || mNfcAdapter == null) --> NOT_SUPPORTED
2. OS-level NFC setting is OFF --> (!mNfcAdapter.isEnabled()) --> NOT_READABLE (pop up a message to turn ON NFC from system settings)

Scenario: 
Open GooglePay with NFC off, the messageBox "Turn on NFC"

Thoughts ?

kenchris, 2019-06-27:
Same is possible with Web Bluetooth - and I don't believe that case is handled. I wonder what @reillyeon thinks

reillyeon, 2019-06-29:
@odejesush has been working on making sure that Bluetooth availability information is available to applications with the `getAvailability()` method. The distinction is that the Bluetooth permission dialog is an opportunity for the user agent to provide a link to system settings to enable Bluetooth while there is no equivalent for NFC and so it is up to web content to detect that NFC is available but disabled.

It seems reasonable to differentiate between the two conditions and return a "not readable" error when NFC hardware is available but disabled and a "not supported" error if it is not present or inoperable. If a user agent decides to display UI when a page is requesting a read operation it could forgo returning the "not readable" error and instead prompt the user to enable NFC itself.

closed 2019-08-13 by kenchris