w3c-cg/web-nfc #238: Allow lower level I/O

kenchris, 2019-06-27:
In the future we want to allow lower level I/O - some exploration

```webidl
dictionary TagMetadata {
  DOMString tagType;
  number transceiveMaxByteLength;
}

dictionary TagFMetadata : TagMetadata {
  UInt8Array manufacturer;
  UInt8Array systemCode;
} 
...

interface NFCReadingEvent : Event {
  readonly attribute DOMString serialNumber;
  readonly attribute NDEFMessage ndef;
  readonly attribute TagMetadata metadata;

  void open();
  void close();
  Promise<Uint8Array> transceive(Uint8Array data);
};

```

jtibble, 2020-02-24:
I would personally like to see this added to the spec and implemented in Chrome. My use-case is that pre-written cards are provided to my organization that do not conform to the NDEF spec, and my coworkers have written custom hex-parsing logic to extract meaningful data from these cards. I'd like to be able to do the same via the web-nfc API without needing an external USB device or using the native Android APIs.

OLmmyT, 2020-03-03:
STMicroelectronics sees many benefits and use cases for this low level API giving access to the transceive() function. It would enable many IoT use cases like:

- Setting and configuring an IoT device, 
- Setting and reading sensors, 
- Personalizing a device before use,
- Developing home automation applications,
- Updating a device’s firmware 
- Adding security (passwords, crypto)

smlu, 2020-04-10:
Any update on this? 
We're working on some web service which also uses ISO/IEC 7816 identification cards and communicates via NFC ISO/IEC 14443 interface to identify users. If WebNFC supportrd ISO depended operations in the future we could ommit building extra platform specifict applications and instead identify users directly via web app in the web browser.

Astriel, 2020-04-28:
Any update on this one ? I've tried the Google Chrome Web NFC Demo, and unfortunatly I'm not able to get informations of the card.

pawisoon, 2021-04-01:
Is there any ongoing work to bring the support for sending and receiving data to ISO/IEC 14443 cards? I see there's many people interested in this kind of feature.

jmelhus, 2024-02-14:
any news about this?

anssiko, 2024-02-14:
Thanks for your interest. We're looking for interested contributors who could pick up the implementation work but cannot commit to any timeline at this time.

If someone lurking on this repo wants to help, please chime in here.

fbrusch, 2024-03-03:
Hi Anssiko, how to get started on this (to give some help?)

anssiko, 2024-03-04:
For concrete implementation contributions, a good starting point is to check out the Chromium code and build it on your target platform: https://www.chromium.org/developers/how-tos/get-the-code/

Other ways to help include e.g. documenting the use cases for the proposed feature to the help bump the priority if this.