w3c-cg/web-nfc #239: Should we just ignore the Abort signal and wait for push response for cannot cancel case?

DonnaWuDongxia, 2019-07-02:
In the SPEC, step 8.2 in the [push method](https://w3c.github.io/web-nfc/#the-push-method):

"If the instance has already initiated NFC data transfer, reject p with "NoModificationAllowedError" DOMException and abort these steps."

If we fail to cancel the push request,  push request may succeed or fail with a different error. If we just reject the promise with  a "NoModificationAllowedError" exception, the developer may lose detailed imformation.  So shall we just rely on the push response if the the operation cannot be canceled?
Actually, for currently case(the NfcImpl.java on Android), if we fail to cancel the push request, the push response will always be called before cancel response. So "NoModificationAllowedError" exception should never occur.

DonnaWuDongxia, 2019-07-03:
Discussions during code review is here: https://chromium-review.googlesource.com/c/chromium/src/+/1680418/1/third_party/blink/renderer/modules/nfc/nfc_writer.cc#168

zolkis, 2019-07-04:
If the implementation knows exactly what happened, the API should allow that information reach the developer. (BTW I am on holiday this week and mostly out of internet reach).

commit 69a3abab (2019-07-25): Ignore abort signal if push operation can't be aborted, fixes #239.
closed 2019-07-25 by kenchris