w3c-cg/web-nfc #242: android.nfc.NfcAdapter.enableReaderMode() has no failure cases

leonhsl, 2019-07-29:
https://w3c.github.io/web-nfc/#dom-nfcreader-start step 3.2 says 
> make a request to all NFC adapters to listen to NDEF messages

and step 3.3 says 
> If the request fails, then
Let e be the result of creating a "NotSupportedError" DOMException.

android.nfc.NfcAdapter.enableReaderMode() [1] does what step 3.2 says, but this function always succeeds in registering a ReaderCallback for receiving notifications from the platform, it has no failure cases, so what does step 3.3 mean for 'If the request fails'?

[1] https://developer.android.com/reference/android/nfc/NfcAdapter.html#enableReaderMode(android.app.Activity,%20android.nfc.NfcAdapter.ReaderCallback,%20int,%20android.os.Bundle)

kenchris, 2019-07-29:
That might not be the case for other platforms

Also, what happens if the Android device doesn't even support NFC? Or NFC is disabled?

leonhsl, 2019-07-31:
We (Chromium impl) will fire the error event NotReadableError in case that NFC is not supported or is disabled, as described by step 2 of https://w3c.github.io/web-nfc/#dom-nfcreader-start.
The used APIs are android.content.Context.GetSystemService(Context.NFC_SERVICE) and android.nfc.NfcAdapter.isEnabled().

kenchris, 2019-08-02:
OK, let's fix the spec then

PR #271 (2019-08-05): Do not force to throw exception when request NFC adapter listening fails
Not all UA will throw error if the request fails for the following step:

> If this is the first listener being set up,
then make a request to all NFC adapters to listen to NDEF messages.

Fixed #242

***
<a href="https://pr-preview.s3.amazonaws.com/Honry/web-nfc/pull/271.html" title="Last updated on Aug 5, 2019, 7:24 AM UTC (ffd38a2)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/271/8220627...Honry:ffd38a2.html" title="Last updated on Aug 5, 2019, 7:24 AM UTC (ffd38a2)">Diff</a>
commit 1675af9a (2019-08-05): Do not force to throw exception when request NFC adapter listening fails

Not all UA will throw error if the request fails for the following step:

> If this is the first listener being set up,
then make a request to all NFC adapters to listen to NDEF messages.

Fixed #242
closed 2019-08-05 by kenchris