w3c-cg/web-nfc #245: Smart Poster Well-Known doesn't seem to actually store a URL itself

kenchris, 2019-07-30:
Instead it just indicates that the tag is to be considered a SmartPoster and additional records will fill in the data, thus it must contain a URL one, but can also contain an additional text record to give description and even records to show images etc.

We need to understand this better and improve the handling, testing and spec

[image]

https://www.oreilly.com/library/view/beginning-nfc/9781449324094/ch04.html#table-appresults

kenchris, 2019-07-30:
[image]

kenchris, 2019-07-30:
Blogpost showing how to read from Android: http://mynewandroidtutorial.blogspot.com/2015/03/nfc-android-read-ndef-tag-textlink.html

NDEF data can span multiple records as indicated in the header. We probably need to check very well how this is handled by the spec

kenchris, 2019-07-30:
https://github.com/w3c/web-nfc/issues/249 has an example showing that this is using composed records within the payload field

kenchris, 2019-08-02:
My suggestion is that we change the type for Smart Posters to `"smart-poster"` instead of `"url"` today. That is a first step.

After that we can potentially create a `toRecords()` getter

Honry, 2019-08-09:
A smart poster record looks like this:
[image]

Honry, 2019-08-09:
> NDEF data can span multiple records as indicated in the header. 

@kenchris, so the CF field should be 1 for a smart poster record?

kenchris, 2019-08-09:
No, these are embedded records! Records inside the payload field of the So record

Honry, 2019-08-09:
Aha, got it.

kenchris, 2019-08-09:
It can actually contain a lot of records, like multiple text ones (for different languages), image data (not sure how they store that) and even signature records (another NFC spec) adding additional security (like you don't want people to set up alternative stickers at a movie teater pointing to a phishing site)

kenchris, 2019-08-09:
91 01 1B 55 ...

> 0x91.toString(2)
'10010001' message begin, no chunked flag, small record, well-known
> 0x01 
type length 1
>0x1B
payload length 11 (google.comQ) - not sure why this ends with Q
> 0x55
"U"
>0x02
https://www. 
>0x67 (g) 6F (o) 6F (o) 67 (g) 6C (l) 65 (e) 2e (.) 63(c) 6f(o) 6d (m) 51 (Q) - google.comQ

kenchris, 2019-08-09:
[0x51.toString(2), 0x01.toString(2), 0x0b.toString(2), String.fromCharCode(0x54), 0x02.toString(2), String.fromCharCode(0x7a), String.fromCharCode(0x68)]

(7) ["1010001", "1", "1011", "T", "10", "z", "h"]

begin, chunked record, well-known
type length 1
payload length 11
type T (0x54 / 1010100)
UTF8 encoding, language length 2 (00000010)
'zh'
'my title'

I don't get though why it is chunked... there are no more chunks... and why not "message end"?

Maybe that app just wrote it wrong and is still able to read it

kenchris, 2019-09-03:
Example of a complex Smart Poster record with sub records.

[image]

[image]

closed 2019-10-23 by kenchris