w3c-cg/web-nfc #261: respec issues

kenchris, 2019-08-02:
@honry @marcoscaceres

how do I link to something like "typed-array-types" in web idl, as well as "unrestricted-double"

`{{unrestricted double}}` doesn't work, but `{{double}}` does

Honry, 2019-08-02:
> {{unrestricted double}}

I've encountered this too, we can find it in xref UI, https://respec.org/xref/,
maybe it's a ReSpec issue? Not sure, @marcoscaceres, need your help. :)

marcoscaceres, 2019-08-05:
The "unrestricted double" issue in now fixed in latest release. If you hit other Respec bugs, please make sure to file them here: http://github.com/w3c/respec/ ... there are a few other contributors, so they can also help resolve things without needing to rely on me too much.

Honry, 2019-08-05:
Thanks @marcoscaceres, fixed in https://github.com/w3c/web-nfc/pull/265

Honry, 2019-08-05:
@kenchris, we can close it now.

closed 2019-08-05 by kenchris