w3c-cg/web-nfc #262: Fix type for external type

kenchris, 2019-08-02:
External type is for custom data, so it is quite important to know what external type was encountered.

Currently we just set the type to `opaque` which is quite useless and forces the developer to do unreliable buffer sniffing.

I suggest we expose the URN type as the type instead like `urn:nfc:ext:intel.com:customrecord`

It is currently an enum so not sure how to do this spec wise

@Honry

kenchris, 2019-08-05:
@marcoscaceres do you have ideas?

marcoscaceres, 2019-08-05:
Urns ⚱️ no es bueno. Just use a string and a registry?

kenchris, 2019-08-05:
But URNs are part of the NFC spec so no way of getting around them :-)

marcoscaceres, 2019-08-05:
I see. Can you give an example of what you are trying to do. I’ve not looked at nfc in any detail.

kenchris, 2019-08-07:
Fixed

closed 2019-08-07 by kenchris