w3c-cg/web-nfc #279: Notify NFCReaders of failures when trying to get a message from a tag?

leonhsl, 2019-08-06:
See
https://developer.android.com/reference/android/nfc/tech/Ndef.html#getNdefMessage().
Once a tag is discovered, we call `android.nfc.tech.Ndef.getNdefMessage()` to try to get a message from it, which may
  - return null if the tag is in the `INITIALIZED` state, i.e. formatted but empty
  - throw exceptions: `TagLostException,` `IOException,` or `FormatException`

In such cases do we need to let all active `NFCReaders` be aware of these failures so that they can choose how to proceed next? i.e. dispatching some error events properly to `NFCReaders.`

Currently http://w3c.github.io/web-nfc/#the-nfc-reading-algorithm only describes the case that a ndef message has already been gotten from the tag, so does the implementation in Chromium.

kenchris, 2019-08-06:
So I assume the problem we have here is that 1) is an empty formatted tag that doesn't even have the empty record?

leonhsl, 2019-08-06:
Exactly.

kenchris, 2019-08-06:
Could we return an empty record?

riju, 2019-08-06:
For context, this [came up in review](https://chromium-review.googlesource.com/c/chromium/src/+/1736167/2/services/device/nfc/android/java/src/org/chromium/device/nfc/NfcImpl.java#561). In chromium implementation, I do see a [kNfcEmpytMessage](https://cs.chromium.org/chromium/src/third_party/blink/renderer/modules/nfc/nfc_constants.cc?g=0&l=40) used to throw an exception while pushing empty message to tag. We could use something similar while reading also ?

kenchris, 2019-08-06:
For TagLostException, we can just avoid dispatching the event as no tag was fully read.

kenchris, 2019-08-06:
We should probably do like Generic Sensor and have an onerror event handler as well, which somehow can handle the error cases

leonhsl, 2019-08-07:
So, for the null ndef message case we dispatch a reading event with a ndef message containing just an empty record; for `IOException` and `FormatException` cases we dispatch an error event. Right?

Note: These events are dispatched to ALL active readers, without any message filtering for matching their reader options.

kenchris, 2019-08-07:
Sounds good, but I think we might want to consider adding a "formatted" type, similarly to the "empty" type

beaufortfrancois, 2019-10-22:
According to https://github.com/w3c/web-nfc/issues/367, reading non-formatted tags result in a onreading event with empty records array for NDEFMessage.
I think we can close this issue right?

beaufortfrancois, 2019-10-22:
Ooops! We should not close it. Dispatching event errors in `IOException` and `FormatException` cases is not supported anymore as we got rid of NDEFErrorEvent.

kenchris, 2019-10-23:
So you suggest we readd the error event for that? Or other solutions?

beaufortfrancois, 2019-10-28:
I didn't find any other solution sadly that satisfies me but re-adding the NDEFError event.
@zolkis for thoughts as well.

zolkis, 2019-10-28:
> I didn't find any other solution sadly that satisfies me but re-adding the NDEFError event.

I think so, too.

beaufortfrancois, 2019-11-05:
For info, I have a PR at https://github.com/w3c/web-nfc/pull/432 that attempts to fix this.

closed 2019-11-05 by zolkis