w3c-cg/web-nfc #304: Attaching "reading" event listener should not be required

beaufortfrancois, 2019-08-22:
I believe attaching an event listener should not have side effects. See https://github.com/w3c/mediasession/issues/148

> In order to receive NFC content, the client needs to attach an event listener for the "reading" event on an NFCReader instance and then activate it by calling NFCReader.scan().

Does that mean `NFCReader.scan()` should not handle NFC content when 'reading' event listener is not attached? If so, we should not use event listeners for this.
If not, we should make sure spec clearly say so.

kenchris, 2019-08-22:
No, as you say reading should have no side effects

kenchris, 2019-08-22:
The text is maybe just not so clear, as you can call scan, and the content will be read but no event handler will receive it as they don't exist... But of course the web page should 'swallow' the content and the Android system shouldnt receive it

beaufortfrancois, 2019-08-22:
Thanks! See https://github.com/w3c/web-nfc/pull/305

closed 2019-08-22 by beaufortfrancois