w3c-cg/web-nfc #308: Type field of Web NFC author records should be "w3.org:A" or "w3.org:a"

leonhsl, 2019-08-27:
The spec[ 2.4.6 Web NFC author records (TNF 4)](https://w3c.github.io/web-nfc/#web-nfc-author-records-tnf-4) says

> Uses NFC Forum external type record with the TYPE field set to "w3.org:A".

But given the fact we found at this [CL](https://chromium-review.googlesource.com/c/chromium/src/+/1769488/2/services/device/nfc/android/java/src/org/chromium/device/nfc/NfcTypeConverter.java#28), 

> NdefRecord record = android.nfc.NdefRecord.createExternal("w3.org", "**A**", bytesOfUrlXXX);
then record.getType() returns just "w3.org:**a**".
we need to figure out where does "w3.org:A" come from. It's from any nfc forum specs? Or the nfc forum specs have any relevant restrictions/clarification about the external record's RTD type?

kenchris, 2019-08-27:
Authors are allowed to use types starting with lowercase, uppercase and even numbers, but more or less standardized types should be uppercase. As this is defined by our spec as a standard and an extension to the NFC specs, using uppercase is most appropriate.

It might be that Android doesn't allow us to do this, in which case we need to reconsider.

Have you checked what it actually writes to the NFC tag? It might be that the getType somehow calls toLowercase or similar

beaufortfrancois, 2019-08-27:
It doesn't look like `getType` does something similar to `toLowercase` according to https://github.com/aosp-mirror/platform_frameworks_base/blob/master/core/java/android/nfc/NdefRecord.java

beaufortfrancois, 2019-08-27:
According to https://developer.android.com/reference/android/nfc/NdefRecord.html#createExternal(java.lang.String,%20java.lang.String,%20byte[])

> NFC Forum requires that the domain and type used in an external record are treated as case insensitive, however Android intent filtering is always case sensitive. **So this method will force the domain and type to lower-case before creating the NDEF Record.**

beaufortfrancois, 2019-08-27:
See https://github.com/aosp-mirror/platform_frameworks_base/blob/master/core/java/android/nfc/NdefRecord.java#L464

```java

    public static NdefRecord createExternal(String domain, String type, byte[] data) {
        if (domain == null) throw new NullPointerException("domain is null");
        if (type == null) throw new NullPointerException("type is null");

        domain = domain.trim().toLowerCase(Locale.ROOT);
        type = type.trim().toLowerCase(Locale.ROOT);
```

kenchris, 2019-08-27:
> NFC Forum requires that the domain and type used in an external record are treated as case insensitive, however Android intent filtering is always case sensitive. **So this method will force the domain and type to lower-case before creating the NDEF Record.**

So that is wrong :(

[image]

kenchris, 2019-08-27:
So yes you need to compare while ignoring the case, so Android might have issues reading external records not written by Android itself.

[image]

But hey we could call those methods outselves and write it upper case - the question is whether we need to use the comparison done by Android?

I guess we can go with lowercase, but we need extra spec text then. Either we lowercase automatically when people try creating an external type with uppercase chars or we force the type to be lowercase (throw or similar)

leonhsl, 2019-08-28:
As nfc forum spec says the domain and type should be treated as case insensitive, web nfc spec using "w3.org:A" does not look worse than "w3.org:a", let's keep with "w3.org:A".

In respect of impl, on Android "w3.org:A" will be written into tags as "w3.org:a", on other platforms (in the future) it may still be "w3.org:A", or it'd be written as "W3.oRg:a" via some other non-webnfc routes, anyway, I'd make Chromium webnfc impl compare them on a case insensitive basis so everything is OK.

I don't think we need to force the type to be lowercase, also webnfc spec does not need to let users be aware that an external custom-type with uppercase chars will always be converted to lowercase for writing, it's an impl detail, users just do what they want based on the knowledge that the external custom-type is case insensitive.

zolkis, 2019-08-29:
Clarified the text in #314

zolkis, 2019-08-29:
@leonhsl Please check if this can be closed.

closed 2019-08-30 by leonhsl