w3c-cg/web-nfc #309: A record might have no payload

kenchris, 2019-08-28:
ie PAYLOAD LENGTH field is 0 and the PAYLOAD field is not present. We need to make sure we actually handle that and that our `to*` methods return `undefined` or similar (at least it makes sense that `toJSON` returns that

kenchris, 2019-08-28:
Same, what happens if I write a 0 length ArrayBuffer as opaque... what is written? and do we get the same back when reading such a record and calling toArrayBuffer?

zolkis, 2019-08-29:
IMHO `toJSON()` will never return `undefined` unless we feed `undefined` to it, i.e. when `record.[[PayloadData]]` is `undefined`. 

The result of `JSON.stringify(undefined)` is `undefined`, but `JSON.parse(undefined)` will through (not valid JSON).

In NFC specs the guiding field is the payload length, PAYLOAD might be there or might be missing.

For writes, the current spec should be fine (when feeding `undefined` to writes).
For reads, we should clarify the parse algorithms to be explicit about the use of `undefined`.

kenchris, 2019-08-29:
Try with Symbol('hi') and BigInt()

kenchris, 2019-08-29:
Values that do not have a JSON representation (such as undefined and functions) do not produce a String. Instead they produce the undefined value. In arrays these values are represented as the String null. In objects an unrepresentable value causes the property to be excluded from stringification.

zolkis, 2019-08-29:
Tried to address in #319

zolkis, 2019-08-30:
@kenchris can we close this? or still valid?

kenchris, 2019-08-30:
I don't think so, like look here:

[image]

That field might not even be present

kenchris, 2019-08-30:
Also here we never set the `[[PayloadData]]`:

[image]

kenchris, 2019-08-30:
We should probably also factor that out in its own algo like the others

zolkis, 2019-08-30:
Right, I have updated only the steps that use the PAYLOAD bytes.

The ones (quoted above) that just attribute PAYLOAD, I didn't change since if PAYLOAD is not present (i.e. undefined), I assumed the [[PatloadData]] will also be `undefined`. Or should this be explicitly specified (for undefined, or null?).

Factoring out in a new algo makes sense.

kenchris, 2019-08-30:
I think it is better if we spell it out, also we cannot say "set to the byte sequence of the PAYLOAD field" if that doesn't exist. As am implemented I would just assume that it alwasy exists

kenchris, 2019-09-30:
Did you make any of these change @zolkis?

zolkis, 2019-09-30:
Not (yet)

zolkis, 2019-10-03:
Seems to be fixed by #323

beaufortfrancois, 2019-10-28:
If we're good, can we close this issue?

closed 2019-10-28 by zolkis