w3c-cg/web-nfc #310: toJSON should return any (which includes undefined)

kenchris, 2019-08-28:
`toJSON` for different objects in the web platform returns different things, some like `performance.toJSON` knows that it will always return structured data (dictionary) and thus returns `object`

Just like the fetch spec we might return anything as and should change the signature to `[NewObject] any toJSON()`

Fetch uses `[NewObject] Promise<any> json();` for comparison

kenchris, 2019-08-28:
Similarly fetch would just return USVString (empty string) and ArrayBuffer (with no associated buffer)

kenchris, 2019-08-28:
I see now that returning `null` is for the case of the empty record. I guess we could add a note for that

zolkis, 2019-08-29:
In our case we might use `toJSON()` for reads and writes. 
In both cases, IMHO `null` maps well. 
On reads, payload length 0 (empty PAYLOAD) will result in `null`.
On writes, `null` will translate to empty PAYLOAD and payload length 0.
But _in principle_, `toJSON()` returns `any` because it could return any simple or compound type or `null`. It would be nice to have Web IDL support for this!

closed 2019-08-30 by riju