w3c-cg/web-nfc #311: Should NFCReader scan() method throw an AbortError ?

riju, 2019-08-28:
[Scan method](url) Step 

> 6. If reader.[[Signal]]’s aborted flag is set, then return.

Do we want to throw an AbortError ? like the [push()](https://w3c.github.io/web-nfc/#aborting-pending-push-operation) Step 7.

zolkis, 2019-08-28:
No, the recommendation is the current text:
https://dom.spec.whatwg.org/#abortsignal-aborted-flag 

This condition happens when the script aborts the scan. Then we don't need to also throw IMHO, but YMMV.

zolkis, 2019-08-28:
Actually what we need to do IMHO is clean up the spec about aborting a push. We need to define aborting a push with the same prose as aborting a scan, and define push cleanup steps for what is aborting a push steps now.

The scan seems OK for me since you get an event onabort and that's enough for a script.

reillyeon, 2019-08-28:
Using the AbortController is the only way to stop a scan and isn't an exceptional case so I don't think we need to dispatch an error on abort.

riju, 2019-08-29:
Thanks @reillyeon and @zolkis, closing this.

closed 2019-08-29 by riju