w3c-cg/web-nfc #312: Support reading embedded records: Add toRecords()

kenchris, 2019-08-29:
NDEF records can have embedded records, and actually the well-known Sp (Smart Poster) makes use of that.

Some example of how to read that on Android here: http://mynewandroidtutorial.blogspot.com/2015/03/nfc-android-read-ndef-tag-textlink.html

We should basically add a `FrozenArray<NDEFRecord> toRecords();` or similar.

Developers are allowed to store their own record types within a subrecord in the well-known type, with the restriction that the type name cannot start with an uppercase (lowercase or number thus is fine).

We should think a bit about how to expose these. Potentially we could create a "custom" recordType or similar, and then add a `customType` property, or we should just add it to the recordType, like "custom:userDefinedType001". We don't need to support this initially, but it is something to think about.

zolkis, 2019-08-29:
If this is additive, I'd wait until the need for the use case in web pages is confirmed by developers.

Especially when I feel the Web NFC spec is getting more and more complex. :)

kenchris, 2019-08-29:
It is definitely needed if you care about storing data more efficiently. I just had people ask about this at BrazilJS conference. Also it is required for smart-poster and it actually doesn't look that hard to implement.

zolkis, 2019-08-29:
So you already got the developer need. Anyway it should be additive and not break the existing API / structures, so we need to figure that out now.

riju, 2019-08-30:
This related to the [smart poster discussions](https://github.com/w3c/web-nfc/issues/245) ?

PR #333 (2019-09-03): Spec toRecords() method
Fixes #312 

I verified this with the NFC specs and it all looks correct. This would work for properly written smart-poster records as well.

***
<a href="https://pr-preview.s3.amazonaws.com/kenchris/web-nfc/pull/333.html" title="Last updated on Sep 3, 2019, 10:37 AM UTC (b65af96)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/333/bfabe0b...kenchris:b65af96.html" title="Last updated on Sep 3, 2019, 10:37 AM UTC (b65af96)">Diff</a>
closed 2019-09-03 by kenchris