w3c-cg/web-nfc #321: Investigate if/how CompressStream/DecompressStream can be used with Web NFC

kenchris, 2019-08-30:
https://github.com/ricea/compressstream-explainer/blob/master/README.md

Maybe just write some examples? Maybe we need some convenience methods etc...

kenchris, 2019-08-30:
It would be nice to easily compress data we are writing to tags but an async streams doesn't work for us as all data needs to be ready when the write happens.

It is probably going to be something like

```
let data = ...
const cs = new CompressStream();
const readable = new Blob(data).stream().pipeThrough(cs);

 while (true) {
    const { value, done } = await reader.read();
    if (done)
      break;
    out.push(value);
    totalSize += value.byteLength;
  }
  const concatenated = new Uint8Array(totalSize);
  let offset = 0;
  for (const array of out) {
    concatenated.set(array, offset);
    offset += array.byteLength;
  }
  return concatenated;
```

taralx, 2020-02-28:
FYI you can do this:

```js
function compressAsync(data) {
  const stream = new Blob(data).stream();
  const compressed = stream.pipeThrough(new CompressionStream('deflate'));
  return new Response(compressed).arrayBuffer();
}

kenchris, 2020-05-20:
Looks like we are good - thanks!

closed 2020-05-20 by kenchris