w3c-cg/web-nfc #324: Missing descriptions on pushing "external type" records

leonhsl, 2019-09-02:
Here: https://w3c.github.io/web-nfc/#creating-web-nfc-message

@kenchris @zolkis @riju

zolkis, 2019-09-02:
We don't use external type in NFCRecortdType, see https://w3c.github.io/web-nfc/#data-mapping.

In turn, we miss/have incomplete definition for how to handle "smart-poster".

zolkis, 2019-09-02:
We don't use external type in NFCRecordType in a clear way, we have a "default" string that has to match a pattern - I don't like that.

We also miss/have incomplete definition for how to handle "smart-poster".

zolkis, 2019-09-02:
This part has changed quite a lot while I was inactive as editor, now wondering about a few other things as well. We used to have a `"web-nfc"` record type, too, a kind of special external record, also used in compatibility checks. It was one of the primary motivations of the Web NFC spec over the low level older web NFC spec. Maybe it has been discussed and dropped, I just need to dig up why.

@kenchris could you check this, handling external type and smart-poster? 
(I could do that only tomorrow).

commit dcdaab42 (2019-09-02): Add algo for mapping external data to ndef, fixes #324
closed 2019-09-02 by kenchris