w3c-cg/web-nfc #327: How does NFCScanOptions#recordType filter external type records?

leonhsl, 2019-09-02:
When reading ndef messages, do we want to match `NFCScanOptions#recordType` exactly for external type records?
By reading the current spec, for `NFCScanOptions#recordType` being 'abc.com:xyz', we only filter those ndef messages containing at least one external type record with type field being exactly 'abc.com:xyz'. 

That is to say, we can filter all 'text' records by specifying `NFCScanOptions#recordType` as 'text', so do 'url', 'json', and 'opaque', but we can not filter all external type records.

I want to confirm whether this is expected behavior for matching external type records.

kenchris, 2019-09-02:
We should be able to filter for external type records, like "w3.org" or "w3.org:a"

leonhsl, 2019-09-16:
This relates closely to #331 but aims to clarify how could NFCScanOptions#recordType match external type records.

Note: `w3.org:*` filter is not possible because `*` is a valid character for the custom-type.

> DIGIT                = %x30-39
ALPHA                = %x41-5A / %x61-7A   ; A-Z / a-z
other                = "(" / ")" / "+" / "," / "-" / ":" / "=" /
                       "@" / ";" / "$" / "_" / "!" / "*" / "'" / "."

kenchris, 2019-09-16:
We will need to allow escaping it like the service worker stuff is doing, but maybe we can ignore it for now, implementation wise.

https://github.com/wanderview/service-worker-scope-pattern-matching/blob/master/explainer.md

I need to look at this when I get back from Japan

kenchris, 2019-09-30:
We should support escaping * with \ for now. 

> The \ character escapes the following pattern character.

kenchris, 2020-05-20:
Going to remove filtering

closed 2020-05-20 by kenchris