w3c-cg/web-nfc #330: onreading event does not fire in Android 6, although it is OK on Android 8

boatbodger, 2019-09-02:
With the javascript below in a web page using Canary 78.0.3900.0, I am finding that on an Android 8 device, the onreading event is firing as expected when a card is placed near the reader.

However, the same code with the same browser version on a Nexus 7, Android 6.0.1 device does NOT produce the onreading event, even though the 'detected' audio is being played (provided Android Beam is 'off')

function readTagId(){
//const writer = new NFCWriter();
const reader = new NFCReader();
	reader.onreading = nfcRead;
	reader.onerror = nfcError;
	reader.scan({compatibility: "any"});
}

function nfcRead(event){
	document.getElementById('datacap').style.visibility='visible';
	document.getElementById('tagId').value = event.serialNumber;
}

function nfcError(er){
	alert("oops...");
	alert(er);
}

boatbodger, 2019-09-04:
Additional info: Problem still present in 78.3901.0.
Also, I can confirm that the tablet gives a different sound if the card is read before scan() is called compared with once scan() has been called.  This confirms that the scan() call is influencing the way the OS is handling the card presence - its just that the event is not being passed back up the stack to the browser event listener, so far as I can tell.

Honry, 2019-09-05:
@boatbodger, I tried on following two envs, both NFC read and write worked well.  
- Nexus 5 Android 6.0.1 Chrome Canary 78.0.3901.0
- Pixel 2 XL Android 9 Chrome Canary 78.0.3901.0 

And this is an implementation specific issue, please report to https://bugs.chromium.org/p/chromium/issues/entry

closed 2019-09-30 by kenchris