w3c-cg/web-nfc #331: "typedef DOMString NDEFRecordType" is expected to be case sensitive or not?

leonhsl, 2019-09-03:
I feel the record type string should always be treated as case insensitive, 
  External type strings: of course we treat them as case insensitive, based on nfc forum spec.
  other type strings:  '`TExt`', '`teXT`'  ==> '`text`', '`JSON`', '`Json`', '`jSoN`' ==> '`json`', etc.

WDYT? If it makes sense to you, we'd better put a sentence describing the 'case insensitive' thing around https://w3c.github.io/web-nfc/#the-ndefrecordtype-string.

Then, for simple impl in Chromium, I'd consider transforming the input NDEFRecordType (`NDEFRecordInit#recordType`, `NFCScanOptions#recordType`) to lower case in the first place.

kenchris, 2019-09-03:
Web APIs are case sensitive

kenchris, 2019-09-03:
[image]

We should be able to provide the URL as on the left side in this table and use that algorithm to derive the right side

leonhsl, 2019-09-03:
Got it, first I'd make `NDEFRecord` ctor throw a TypeError in case of the record type being provided as '`TeXT`' alike.

leonhsl, 2019-09-09:
Let me take an example to ask the question:
If users create an external record with NDEFRecordInit#recordType being '`www.example.com:xyz`', for pushing, do you mean that we need to transform the type to '`example.com:xyz`'? i.e. using its registrable domain instead?

leonhsl, 2019-09-09:
If the answer is yes, `null` registrable domain would be a problem, since Android SDK says [1] that both domain and type should be non-empty.

> The unchecked exception IllegalArgumentException will be thrown if the domain and type have serious problems, for example if either field is empty, so always catch this exception if you are passing user-generated data into this method.

[1] https://developer.android.com/reference/android/nfc/NdefRecord.html#createExternal(java.lang.String,%20java.lang.String,%20byte%5B%5D)

kenchris, 2019-09-09:
We just shouldn't allow those, like github.io which is registered... this means that I can use kenchris.github.io:shoppingitem

leonhsl, 2019-09-10:
`xyz.github.io`'s registrable domain is still the same `xyz.github.io`, so it does not matter on the decision whether we transform the type to its registrable domain or not.

My question was: do we need to transform 'www.example.com:xyz' to 'example.com:xyz'?

And another new question: from your last comment, it seems to me that the impl should throw a TypeError for the user input `github.io` because it's an already registered one?

The spec needs to be clear about this.

kenchris, 2019-09-10:
Yes, because that is the registered domain, and it is common that companies use a lot of subdomains like blah.intel.com and phone.intel.com etc and they should all be registered as intel.com. Many sites also work whether you add www. or not and it would be bad if I wrote a tag when on www. and then not being able to read it when I didn't add www.

kenchris, 2019-09-10:
I am totally ok with adding the TypeError exception

reillyeon, 2019-09-11:
Are we sure that we want to rely on the Public Suffix List in this specification?

https://github.com/sleevi/psl-problems

kenchris, 2019-09-16:
@zoltan I dont think that the ID field compresses the URLs like the url records but I might be wrong.

Can you make the spec changes and add good examples showing the suggestions that @sleevi made?

For the external records, the NFC specs does have some text saying to basically use the registered domain and not fully qualified domain names, but we can just add that as a note because I dont see the NFC spec disallowing people from doing whatever they want to do

kenchris, 2019-09-16:
> The NDEF Record Type Definition spec says
> 
> > Two External Type Names MUST be compared in a **case-insensitive** manner. Example:
> > “example.com:foobar”
> > “Example.com:foobar”
> > “Example.COM:Foobar”
> > “eXaMpLe.CoM:fOoBaR”
> > The four examples above represent all the same External Type Name.

Meaning that we can allow only lower case values and can expose the value as lower case only.

leonhsl, 2019-09-16:
So I'll make our impl go with FQDN.

About this, by my understanding of the **case-insensitive** thing described in NFC spec, should we just allow all lower and higher case characters and leave users to care about the case comparision things? Can you clarify more for Web NFC spec to add such a constraint?
 
> Meaning that we can allow only lower case values and can expose the value as lower case only.

kenchris, 2019-09-16:
- We should only allow lowercase input
- When we read from tags, we convert to lower case so all comparisons will be on lowercase values
- When we expose the values (output) they are also lowercased

leonhsl, 2019-09-16:
Sorry I was asking why... Is it due to some special reason for adding such an extra constraint?

leonhsl, 2019-09-16:
Except for `The NDEFRecordType string`, there're other places also using `registrable domain` in Web NFC spec, do we need to rethink about them with the PSL problem we're now clear of?

kenchris, 2019-09-16:
I don't see any reason to write the url as anything but lowercase when it is compared lower case as well - that is totally allowed by the spec.

The value is very similar to an enum but isn't an enum because you can add urls for external record types, and the default naming guideline from the TAG is that such enums should be lowercase:

https://w3ctag.github.io/design-principles/#casing-rules

zolkis, 2019-09-16:
Small recap. NFC specs says comparisons should be done case-insensitive.

Therefore apps can use mixed case without issues, there is no constraint from NFC.

However, the Web NFC spec (as an NFC _app_) can impose policies like:
1. We allow writing only FQDN and be specific in the spec on how we do comparisons and what is left to applications to figure out about how to specify filtering URL patterns.
2. We lowercase the input from NDEF messages we read, use that in comparisons and present that to the web page/apps.
3. We don't lowercase input from NDEF messages we read, use case-insensitive comparison where the NDEF specs dictate, but expose the unchanged content to web pages/apps.
4. We only accept lowercase input from web pages.
5. We silently convert input from web pages to lowercase.
6. We write input from web pages as it is (consistent with 1).
(7. We can use encoding that is best for the web, an issue discussed earlier.)

Note that out of the above, Kenneth  2+4.

My own opinion: 
- for policy 1, we got useful input from @sleevi (thanks!) that we must have this (with reasons/examples to be exposed in the spec).
- for 2+4, [suggested](https://github.com/w3c/web-nfc/issues/331#issuecomment-531616728) by Kenneth, I am neutral, but if anything, I find it slightly inconsistent, since if we silently lowercase output, why not _silently_ lowercase input as well (policy 5)? So 2+5 would be more consistent. This might simplify things for implementation and developers who don't read the spec, just intuitively want to do something with Web NFC API as it is, write mixed case to tags and wonder why do they get a match when they (for instance) have expected case-sensitive comparisons. 2+5 would fix that, they'd soon realize that we are the knights who say _lowercase_ :).

The alternative to 2+4 (or 2+5) would be case-transparency (3+6): only the comparisons done case-insensitive where the NDEF specs require it, and we are clear about this in the spec, but we don't change content.

IMHO 1+3+6 would be a more consistent set of policies, so there my vote goes. I am open to corrections/suggestions.

kenchris, 2019-09-16:
We could lowercase the urls, but I don't want to accept the other values like "JsOn"

leonhsl, 2019-09-16:
+1 to 1+3+6

I think we're still discussing the external type string, right? First, it does not involve with any **urls**, second, `example.com:JsOn` should be OK, no matter we go with 2+4 or 3+6, but of course `JsOn` is **NOT** OK, because it's not equal to `json` which is a well-known record type (not external record type) and should be always treated in a case-sensitive way from a previous comment.

> Web APIs are case sensitive, thus "TeXT" should be rejected as it is not equal to "text".

leonhsl, 2019-10-10:
Could we move forward here to define a clear/exact naming rule for external types in spec? We want to implement based on the correct definition to address any potential security concerns.

leonhsl, 2019-10-10:
#350 is also focused on the naming rules for external types.

zolkis, 2019-10-10:
OK, I propose we move on with 1+3+6 that you okayed as well.

>  - Allow writing only FQDN and be specific in the spec on how we do comparisons and what is left to applications to figure out about how to specify filtering URL patterns.
> - Don't lowercase input from NDEF messages we read, use case-insensitive comparison where the NDEF specs dictate, but expose the unchanged content to web pages/apps.
> - Write input from web pages as it comes (no silent lowercase conversion).

Is that OK? Spec changes come later but you can assume the above for the impl (if you agree).

kenchris, 2019-10-18:
Can we close?

sleevi, 2019-10-18:
I'm not sure who that is directed to. I'm used to seeing folks pinged via handle if there's a specific question, and usually see spec text to accompany the proposal to ensure it's clear and unambiguous.

If it was for me, https://github.com/w3c/web-nfc/issues/331#issuecomment-540402533 is not clear to me, and spec changes would be easier for me to make positive contributions. It's hard for me to fathom the implications of the proposal otherwise, but that may be the fact that I'm not working on this, and was just trying to highlight the issues that come up and the impact they might have on the platform.

beaufortfrancois, 2019-10-28:
@zolkis Can you start a spec PR with https://github.com/w3c/web-nfc/issues/331#issuecomment-540402533 changes?

zolkis, 2019-10-28:
> Can you start a spec PR with #331 (comment) changes? 

That's already done in #373 (somehow failed to reference it), except that we might need an algorithm for FQDN (or leave that to the implementation, but further improve spec prose about external type validation).

beaufortfrancois, 2019-12-11:
@zolkis What's left there?

zolkis, 2019-12-11:
From my part the validate external/local types algorithms address this (to be checked), but we need to fix #350 that is related.

zolkis, 2019-12-17:
I think this has been fixed by #474 (and #477).
@leonhsl could you please check if we can close this?

leonhsl, 2019-12-18:
Cool, the algorithms for validating external/local type names look pretty neat now. Thanks @zolkis !

closed 2019-12-18 by leonhsl

zolkis, 2019-12-18:
And thanks @kenchris for the comments/help/tips/work on the external type validation.