w3c-cg/web-nfc #336: Make sure we support sub-records AKA NDEF message as payload

kenchris, 2019-09-03:
According to the NFC specs, payload may be a NDEF message (meaning a sequence of NDEF records)

For instance from Smart Poster spec: "The content of a Smart Poster payload is an NDEF message. The contents of this message consists of several NDEF records."

* Make sure we can read these sub-records
* Make sure we can read custom records (as defined in RTD)
* Make sure we can write sub-records in external records

zolkis, 2019-09-03:
Let's not introduce a new term "sub-record", the NFC Forum Smart Poster and the NDEF spec just speak about NDEF message and records, including Smart Poster, which says

> The content of a Smart Poster payload is an NDEF message. The contents of this message consist of several NDEF records. The Smart Poster can have zero, one, or more of the following
components: (...) the Title record (...) the URI record (...) the Action record (...) the Icon record (...) the Size record (...) there MAY be other records, which can be treated in an application-specific manner. For example, some application might include a vCard contact card using the proper MIME type. Applications MAY ignore any extra records inside the Smart Poster.

kenchris, 2019-09-03:
OK, that is fine, so you want to change `toRecords()` to `toMessage()` instead? It makes the user code a bit longer and we don't have the `url` member, but I guess we can soon get rid of that with the introduction of id.

zolkis, 2019-09-03:
I have no issue with `toRecords()` per se, just the confusion (need of differentiation) between NDEF and added terms/solutions, i.e. what comes from NDEF, what comes from Web NFC and what comes from an app.

beaufortfrancois, 2019-09-04:
@kenchris Is it possible to push a smart poster NDEF message that contains several records? If so, how?

kenchris, 2019-09-04:
Not yet, but I am getting there

kenchris, 2019-09-30:
@beaufortfrancois that is happening as part of https://github.com/w3c/web-nfc/issues/357

closed 2019-10-18 by kenchris