w3c-cg/web-nfc #342: Fix spec issue after things from promise guide moved into web idl

kenchris, 2019-09-04:
[image]

Either we need to use

[image]

or

[image]

kenchris, 2019-09-04:
Ah should be the former...

zolkis, 2019-09-04:
Anyway we need to specify what happens on fulfillment and rejection. Transforming a promise is before settling, right?

kenchris, 2019-09-04:
This is the text that was removed from promise guide:

"<dfn id="transforming-by">Transforming</dfn> <var>p</var> with a fulfillment and/or rejection handler" is shorthand for calling <code>p.then(fulfillmentHandler, rejectionHandler)</code>, using the initial value of <a href="https://tc39.github.io/ecma262/#sec-promise.prototype.then"><code>Promise.prototype.then</code></a>.

kenchris, 2019-09-04:
Not exactly sure how the new wording should be @domenic

domenic, 2019-09-04:
Sorry about this; we just fixed that with https://github.com/heycam/webidl/pull/792. You can now use "react to _promise_".

closed 2019-09-05 by kenchris