w3c-cg/web-nfc #349: Two 'ignoreRead' relevant descriptions look wrong

Honry, 2019-09-10:
1. Look at the definition of [ignoreRead](https://w3c.github.io/web-nfc/#dom-nfcpushoptions-ignoreread), does it only work for an NFC tag? 

 > When the value of the ignoreRead property is true, the push algorithm will skip invoking the receiving and parsing steps for an NFC tag.

2. In `push` operation, 
> If options's ignoreRead is not equal to true, run the receiving steps.

While in the [receiving steps](https://w3c.github.io/web-nfc/#steps-receiving), “step 3 of the NFC listen algorithm.” is out of sync, what was the exact step it intended to refer to? And hardcoding step number is not appropriate.

> If there are any NFCReader instances in activated reader objects then UAs MUST listen to NDEF messages, according to step 3 of the NFC listen algorithm.

zolkis, 2019-09-16:
> Look at the definition of ignoreRead, does it only work for an NFC tag?

Yes, that's only for tags. 

> according to step 3 of the NFC listen algorithm 

That part of the sentence is not needed, I will remove it.

Honry, 2019-09-17:
> Look at the definition of ignoreRead, does it only work for an NFC tag?
Yes, that's only for tags.

@zolkis, thanks for your confirmation!

commit 92933cfc (2019-09-24): Fix #349: remove unnecessary step number reference
commit faf64037 (2019-09-30): Fix #349: remove unnecessary step number reference