w3c-cg/web-nfc #351: The description of multiple scan() calls is a bit vague

Honry, 2019-09-16:
In [scan() method](https://w3c.github.io/web-nfc/#the-scan-method), spec says: 
>Multiple consecutive calls to the scan() method from the same origin create filters which are in OR relationship.

Multiple calls to the scan() method may have following two scenarios:
1. Multiple scan() calls from different NFCReader objects in the same origin, they create various filters which should be independent. This does not apply to the "OR relationship" I think.

2. Multiple scan() calls from a single NFCReader object. Looks like it could only create one filter for a reader object. Maybe the latter call to the scan() method should override the former one. But this should be explicitly explained in the spec.

Cc/ @leonhsl, @kenchris, @zolkis

zolkis, 2019-09-16:
I think this is a leftover from NFCReader related spec changes done earlier.

There is some inconsistency with NFCReader.
The NFCReader object just gives access to the read API. It is the `scan()` method that takes filters.
However, the NFCReader object has internal slots from a single filter. Looks like repeated calls to `scan()` will replace the filter.
The global slot [[ActivatedReaderList]] takes multiple possible reader objects, each with internal slots initialized from one filter. Note that the abort signal is also put there by the scan() algo.
I wonder could we take the filter to the NFCReader constructor, @kenchris? Or would the abort signal related steps of the [scan()](https://w3c.github.io/web-nfc/#the-scan-method) algo be considered side-effect in a constructor?

It seems we have no ways to remove NFCReader objects from [[ActivatedReaderList]], other than the abort signal, or if that was not set, then it needs invoking scan() with it an abort signal and then abort that. Non-abortable readers cannot be removed otherwise.
BTW step 3.1 in the [scan() algo](https://w3c.github.io/web-nfc/#the-scan-method) doesn't check for `undefined` and a filter with its `signal` set to `undefined` will break the algo in step 7, IMHO.

Since now we have NFCReader objects, essentially bound to a single filter, we currently don't have a way to support combined filters, therefore we should remove that sentence.

It's another issue (needing a use case) whether we want to support combined filters, e.g. 
1. a list of filters that would be in OR relationship, 
2. or at the filter dictionary level we'd allow arrays with OR semantics and on the upper level a list of filters would have AND semantics. 

At the moment we will not support any combined filters in the coming version.

zolkis, 2019-09-16:
@kenchris could you help having a second eye on the scan algorithm for this case:
1. `scan()` is invoked with `options` that contain a `signal`: the internal slots get initialized, content is listened for.
2.  `scan()` is invoked again, with different `options` that _do not_ contain a `signal`. The rest of `options` replaces the internal slots, but what happens to the abort signal? Will the previously set value stay (yes, if I read the algo correctly) and is that something we want?

kenchris, 2019-09-30:
In theory, as we can have multiple readers, they filtering should really be independent. I guess we could create another issue for that and treat it as an enhancement

zolkis, 2019-12-17:
@Honry please check if this can be closed. If not, assign me.

Honry, 2019-12-18:
@zolkis, yes, this could be closed now. And I just create new one for filters from multiple readers at https://github.com/w3c/web-nfc/issues/479.

closed 2019-12-18 by Honry