w3c-cg/web-nfc #354: Data Mapping part is missing toRecords()

leonhsl, 2019-09-24:
With the description in https://w3c.github.io/web-nfc/#the-ndefrecord-interface:
> external
If the recordType value is an external type, then return the result of running parse records from bytes on bytes.
Otherwise, return null.

`toRecords()` should be only available for external type records, which should be clear in [Data Mapping](https://w3c.github.io/web-nfc/#data-mapping).

kenchris, 2019-09-30:
It should always be available but might just return null.

It should also return valid values for Sp smart poster and potentially other well-known records

kenchris, 2019-09-30:
The `toRecords` is also not a link, which it should be.

Also, the following won't work for Sp:

[image]

@zolkis

zolkis, 2019-10-03:
Partly fixed by #359 and I added the missing row in Data mapping in #361

kenchris, 2019-10-23:
What is missing @zolkis

zolkis, 2019-10-23:
It should be fine now.

closed 2019-11-04 by kenchris