w3c-cg/web-nfc #357: Do we want to add support for writing sub records?

leonhsl, 2019-09-25:
As we know, a ndef record may contain a ndef message as payload, and web nfc spec already adds support for reading those sub records, i.e. `NDEFRecord#toRecords()`
Then, do we want to support writing as well?

zolkis, 2019-09-25:
Short answer: yes.

Long answer below.

The informative section 4.3.C. of the NFC Forum RTD spec says this is possible, but also that it's _defined by the application_, together with the use of ID as context or other internal referencing method. IOW they don't exclude that NDEF message format could be used in a recursive manner to describe message _trees_.

The Web NFC API being such an "application", we could make the private decision to support parsing and constructing message trees that since it may be convenient for developers.

I think we should reformulate the algo for [`toRecords()`](https://w3c.github.io/web-nfc/#dom-ndefrecord-torecords) to specify since I am not convinced that `toRecords()` should only operate recursively on external records (step 3.1. in the [convert payload](https://w3c.github.io/web-nfc/#dfn-convert-ndefrecord-payloaddata-bytes) steps). According to the RTD at least we can side any number of records and messages, only that we need to encompass them with the right payload length field into the containing record. FWIW "Sp" is well known type so the algo would not work on it (would fail in step 3.1). I understand we just piggyback-invoke the algo to cover the recursivity, but then we should either get rid of the check on TNF (which we don't want since we need to check on [proper ABNF](https://w3c.github.io/web-nfc/#dfn-external-type)), or otherwise reformulate the algo for [`toRecords()`](https://w3c.github.io/web-nfc/#dom-ndefrecord-torecords).

IMHO the API can already express defining a message tree by simply assigning `data` of [NDEFRecord](https://w3c.github.io/web-nfc/#the-ndefrecord-interface) to an [NDEFMessage](https://w3c.github.io/web-nfc/#the-ndefmessage-interface) object for instance.

However, we may want to validate that input (e.g. allow trees only on TNF=1 "Sp" and TNF=4). We could also do that in the [create Web NFC message](https://w3c.github.io/web-nfc/#dfn-create-web-nfc-message) steps and then we don't need Web IDL changes, just algorithm and implementation changes.

zolkis, 2019-09-25:
If you agree with the above (double-check for possible errors/omissions) I could make a PR with those changes.

leonhsl, 2019-09-26:
About `toRecords`(), yeah I agree we need to make our spec clarify that it's only for TNF=1 "Sp" and TNF=4.

About writing,
> IMHO the API can already express defining a message tree by simply assigning data of NDEFRecord to an NDEFMessage object for instance.

Implementation can do that but it would be quite hard for users to prepare a properly formatted raw bytes (NDEFRecord#data) from which we can make out a valid NDEFMessage.
We can also consider adding a new member "`NDEFMessageInit payloadMessage;`" into `NDEFRecordInit`, so users can construct the tree easily, but this makes things complex, Web IDL with recursive things seems so hard to understand, in addition we also need to point out that `payloadMessage` is only for "Sp" and external type records...
I'm unsure whether it's worth doing that, seems more considerations are needed.

zolkis, 2019-09-26:
> Implementation can do that but it would be quite hard for users to prepare a properly formatted raw bytes (NDEFRecord#data) from which we can make out a valid NDEFMessage.

I though the users would just assign `data` to an `NDEFMessage` _object_ and the algo & impl would check if `data` is NDEFMessage (I know that's not trivial, but that's the price), then parse recursively and build the proper byte array by implementation. 

You may be right that we might want to explicitly clarify the intent by adding a new member to `NDEFRecord`, e.g. a boolean meaning "treat payload as NDEFMessage", now we need a good name for that.

leonhsl, 2019-09-26:
Oh I overlooked that `NDEFRecordInit#data` is of `any` type, sure we can feed it with an `NDEFMessage` or `NDEFMessageInit`, but only in case of "Sp" or external type records.

kenchris, 2019-09-26:
@zolkis this sounds like an OK approach. You should share some code samples so that we can better judge?

kenchris, 2019-09-26:
> Oh I overlooked that `NDEFRecordInit#data` is of `any` type, sure we can feed it with an `NDEFMessage` or `NDEFMessageInit`, but only in case of "Sp" or external type records.

People can write whatever data they want in an `opaque` record, but have to deal with how to read it themselves, so I don't see why they should be restricted

leonhsl, 2019-09-26:
I don't know the exact restrictions in nfc forum spec related to payload ndef message, is a TNF=2 (opaque) record allowed to contain a ndef message as payload there? It needs to be clear in web nfc spec accordingly. 

I suggest to first amend web nfc spec to make out a complete logic of reading (`toRecords`()), then add the logic of writing with a separate PR for which we might need more discussion.

kenchris, 2019-09-26:
Want to proceed @zolkis ?

zolkis, 2019-09-26:
TNF=2 can store any data once MIME type is also given. Since NDEFMessage may contain multiple different records with different MIME types, it doesn't make sense to allow message as payload for TNF=2.

However, it could be allowed also for TNF=5 (unknown) as opaque types without MIME type. 

Also, for TNF=6 (unchanged) we should allow message as payload if it's a chunk-record and the first record has been a tree-allowed type ( TNF=1 "Sp" or TNF=4 or TNF=5).

zolkis, 2019-09-26:
Yes, I can make a PR.

zolkis, 2019-09-26:
Hmm, just making the PR and noticed we don't seem to handle TNF=6 in the spec. I will leave it out for the moment and let's track that in a separate issue.

zolkis, 2019-09-30:
Yes, I can add an example.

Question: should we change the parsing steps for Sp, TNF=4,5?
Currently we don't parse the payload, just give the bytes to the apps. We could try to recursively parse the content on these payloads to NDEFMessage and if that fails, we pass on the raw bytes.

kenchris, 2019-09-30:
@zolkis what is the advantage?

We allow toArrayBuffer for all right?

I mean, I already have to call toRecords() so whether we go the parsing at that point or earlier, probably doesn't make much of a difference.

I guess the only problem we potentially have is if we don't parse special well-known records, like the Signature record or other special ones like Bluetooth Handover records etc.

Currently I guess you can only access those using raw bytes, but that is another issue to look into

zolkis, 2019-09-30:
OK, then that's good enough.
Will add an example for Sp and one for nested external.

Shouldn't we split examples in sub-sections so that they appear in the Table of Content?

kenchris, 2019-09-30:
I think that would be a fine thing to do

zolkis, 2019-09-30:
OK, I will do that, too. Now, lunch :)

zolkis, 2019-10-03:
Should be fixed by #359

leonhsl, 2019-10-12:
1. https://w3c.github.io/web-nfc/#data-mapping is saying `toRecords()` is only available to smart poster records. It should be available to external type records (TNF=4) and unknown type records (TNF=5) as well.

2. https://w3c.github.io/web-nfc/#read-an-external-record-with-an-ndef-message-as-payload introduces a new concept `local types` for embedded records, which I think needs to be regulated by https://w3c.github.io/web-nfc/#dom-ndefrecordtype as well.
> For external type records, local types are application specific. Local types are exposed as a recordType, in this example prefixed with "-".

zolkis, 2019-10-12:
> https://w3c.github.io/web-nfc/#data-mapping is saying toRecords() is only available to smart poster records. It should be available to external type records (TNF=4) and unknown type records (TNF=5) as well.

In my interpretation that table was meant to say what mappings should work, not what is available on the object.

Yes, `toRecords()` may return an `NDEFMessage` for TNF=1 "Sp", TNF=4 and TNF=5 as already explained in the [algo](https://w3c.github.io/web-nfc/#dfn-convert-ndefrecord-payloaddata-bytes). 
In all other case it might be available but will return `null` or will throw.

You are right that the data mapping table should be updated, but it should be emphasised that it's not about the presence of a getter, but what is supposed (meant) to work.

> introduces a new concept local types for embedded records, which I think needs to be regulated by https://w3c.github.io/web-nfc/#dom-ndefrecordtype as well.

The local types are defined [here](https://w3c.github.io/web-nfc/#well-known-type-records-tnf-1).

leonhsl, 2019-10-14:
1. The data mapping table needs to be updated for `toRecords`.

> The local types are defined here.
2. This is just the definition of NFC Forum local type, but it should also be captured well at https://w3c.github.io/web-nfc/#dom-ndefrecordtype because it's accessed via `NDEFRecord#recordType` in Javascript by users.
And, the example https://w3c.github.io/web-nfc/#read-an-external-record-with-an-ndef-message-as-payload makes a local type prefixed with '-', which is against  NFC Forum local type definition saying 'always start with lowercase letter or a number'.

leonhsl, 2019-10-14:
3. About writing embedded records, `NDEFRecordInit#data` could be NDEFMessageInit in case of smart poster records, opaque records and external type records, the data mapping table needs to be updated for this, currently it only has smart poster records.

4. The algo https://w3c.github.io/web-nfc/#creating-ndef-message says
> "opaque"
If record's data is NDEFMessageSource, then return the result of running the create NDEF message given record's data.
Otherwise, map binary data to NDEF.

I think it should be `NDEFMessageInit` rather than `NDEFMessageSource`. `NDEFMessageInit` is a union type including `ArrayBuffer`, in which case we should just use the `ArrayBuffer` directly as payload rather than creating another nested message containing this `ArrayBuffer`.

kenchris, 2019-10-14:
> 1. makes a local type prefixed with '-', which is against  NFC Forum local type definition saying 'always start with lowercase letter or a number'.

Yes, we just need to fix that example to now use -

kenchris, 2019-10-14:
> I think it should be `NDEFMessageInit` rather than `NDEFMessageSource`. `NDEFMessageInit` is a union type including `ArrayBuffer`, in which case we should just use the `ArrayBuffer` directly as payload rather than creating another nested message containing this `ArrayBuffer`.

No, people can decide themselves what kind of data they write and one type could be nested records, in which case we transform those records into bytes and write the bytes @zolkis

leonhsl, 2019-10-14:
How can people tell us they want a nested message as payload, given just an `ArrayBuffer`?  It's good enough that `NDEFMessageSource` acts as a convenient way for calling `push()`, but using it to construct nested messages makes things complex.

kenchris, 2019-10-14:
If you give an ArrayBuffer that arraybuffer data will be written by us, we will do no processing or modification of the data.

If people give a NDEFMessageInit (should be union member of MDEFMessageSource) then we write write those records directly (or think of it as we turn that into an arraybuffer and write it)

leonhsl, 2019-10-14:
Yes that's what we expect and the spec is already in the right direction.
The question is, how do you know the input ArrayBuffer is an `ArrayBuffer` or an `NDEFMessageSource` (union type including ArrayBuffer)? That's what I asked about.

kenchris, 2019-10-14:
[image]

(this should have included smart-records as well @zolkis)

Of course it is a bit harder than just using the Android APIs here as you basically have to do the opposite of https://w3c.github.io/web-nfc/#parsing-records-from-bytes

zolkis, 2019-10-14:
Given the recent shift towards low-level API, the current type mapping is more and more dissonant.
I noticed that it's really confusing to people and it's frustrating for me, too.

Therefore I would like to shift to a simple and direct type-mapping, almost the same as now, but more direct mapping with the NFC specs.
```
TNF=0 --> "empty"
TNF=1 --> TYPE="T" --> recordType="text", mediaType="text/plain"
            --> TYPE="U" --> recordType="url", mediaType="text/plain"
            --> TYPE="Sp" --> recordType="smart-poster" (with local types), mediaType=
[
           --> TYPE="Sig" --> recordType="signature", mediaType=TBD
            --> TYPE="Hc"  -->recordType="handover-carrier", mediaType=TBD
            --> TYPE="Hr"  -->recordType="handover-request", mediaType=TBD
            --> TYPE="Hs"  -->recordType="handover-select"
]
TNF=2, TYPE=<mime-type> --> recordType="media" or "mime", mediaType=<mime-type>
TNF=3, TYPE=<url> --> recordType="absolute-url", mediaType=<url>
TNF=4, TYPE=<external-type> --> recordType="external", mediaType=<external-type>
TNF=5, no TYPE --> recordType="unknown", mediaType="application/octet-stream"
TNF=6, CF=1, ID, TYPE=<mime-type>  --> recordType="unchanged", mediaType=<mime-type> (starting chunk)
TNF=6, CF=1, no ID, no TYPE --> recordType="unchanged", mediaType=null (mid chunk)
TNF=6, CF=0, no ID, no TYPE --> recordType="unchanged", mediaType=null (last chunk)
```

We probably won't see TNF=6, but the spec needs to map it.

This is almost the same as we have now, but no `"opaque"` and no `"json"`, though I think it might be useful to allow recordType="json" as a sub-case of "media" (or "mime") for reverse mapping (that would map to TNF=2, TYPE=<json-type> with default value "application/json".

kenchris, 2019-10-14:
Ah I see the mistakes, yes that should probably be *Init and not *Source

kenchris, 2019-10-14:
@zolkis I am not sure we can data with mime type for "mime" or "media" in existing APIs, opaque might not be such a bad name

kenchris, 2019-10-14:
From MIMESNIF: Standards are encouraged to consistently use the term MIME type to avoid confusion with the use of media type as described in Media Queries. [MEDIAQUERIES] - definitely not use `media`

zolkis, 2019-10-14:
Right, the names need some discussion, but we should not use `opaque` for more than one TNF. So if `"mime"` is acceptable, I would like that better, but we could use `"opaque"` for TNF=2.

kenchris, 2019-10-18:
I am closing this for now. Feel free to open more specific issues

closed 2019-10-18 by kenchris