w3c-cg/web-nfc #358: Receiving empty NFC messages since Chrome 77.0.3865.92

richardrobberse, 2019-09-26:
I know this is most likely unrelated to web-nfc itself, but since Chrome version 77.0.3865.92 the watcher receives empty NFC messages.
Is anyone aware of this and/or do you have suggestions for a solution?

Here's a codesandbox (following the spec) to reproduce it:
https://vxufp.csb.app/

First, click the 'Add NFC watcher' button and then scan an NFC tag.

Suggestions are more than welcome!

Honry, 2019-09-26:
Looks like there's a bug in your code, per the old spec at https://cdn.staticaly.com/gh/w3c/web-nfc/336e6afe18470caeca47967413acb39a34c86fcd/index.html#the-nfcmessage-dictionary, in following code snippet, `message.data[0]` is undefined, it should be `message.records[0]` 

```
      if (message.data[0].recordType === "empty") {
        navigator.nfc.push([{
          url: message.url,
          data: [{
            recordType: "text",
            data: "Hello World"
          }]
        }]);
      }
```

richardrobberse, 2019-09-26:
@Honry You're right but this is not the cause of the issue, I just copy-pasted some code in the sandbox to repo it quickly, it's not my production code (luckily)
The thing is, that the message received in the callback is empty since Chrome 77, it's an empty object.

Honry, 2019-09-27:
I got an empty data as well, see my debug screenshot below, looks like `data` attribute is missing in NDEFRecord object. Cc/ @leonhsl 

[image]

leonhsl, 2019-09-27:
`NDEFRecord#data` no longer exists anymore, instead we have `NDEFRecord#toXXX()` data getters. See https://w3c.github.io/web-nfc/#data-mapping.
Also, `navigator.nfc` is the old interface that has already been replaced with `NFC{Reader,Writer}` in Chrome.

Honry, 2019-09-27:
The problem is the latest Chrome stable has not support the new NFC API yet. @richardrobberse, can you transfer your project to Chrome canary with the new NFC API?

leonhsl, 2019-09-27:
Yeah the new APIs are quite better than before :) And, even in the case above, `toText()` could still work.

richardrobberse, 2019-09-27:
@leonhsl I've tried to use the `toText()` still in combination with the old API, but I can't get it to work.
There should still be a way to get it to work with the old API, right? Since the latest version of Chrome doesn't support the new API yet.

leonhsl, 2019-09-27:
Ah I was wrong in the previous comment. 77.0.3865.92 does not have toXXX() getters yet, but it should already be having the new API NFC{Reader,Writer} ready.

richardrobberse, 2019-09-27:
@leonhsl I see, I found [this example](https://w3c.github.io/web-nfc/#example-4-read-data-from-tag-and-write-to-empty-ones) but where's the `NFCReader()` coming from? It's undefined for me.

Honry, 2019-09-27:
Looks like the `NFCReader()` has not been landed in Chrome 77, so strongly suggest you to try new NFC API on latest Chrome canary, you'll get almost full API support there.

richardrobberse, 2019-09-27:
Yeah, I will, but I have customers on an app that requires NFC so canary won't help me out.
Some have received the latest Chrome update :<

leonhsl, 2019-09-27:
As Web NFC API is still in the middle of developing, and the old navigator.nfc was expected to be obsoleted, it's nearly impossible for us to fix any problems for it on a stable version Chrome. Sorry about that.. Chrome 78 and later versions should be OK.

richardrobberse, 2019-09-27:
@leonhsl Thanks a lot for your help and transparency.
So what to expect in Chrome 78?
I just ran my example on Chrome 78 Beta and it seems that `navigator.nfc` is officially deprecated there.
That means I should move to the new API, but the `NFCReader()` is still undefined, is this just a matter of time before it gets implemented?

leonhsl, 2019-09-29:
NFC{Reader,Writer} interfaces should be usable in Chrome 78, but not with the full functionality as described in the spec, we're trying to make a version qualified for Origin Trial these days.
Yes navigator.nfc is completedly removed from 78. And, about the undefined `NFCReader` problem, did you turn on the `WebNFC` flag in chrome://flags?

kenchris, 2019-09-30:
Can we close?

richardrobberse, 2019-09-30:
@kenchris  not yet.
@leonhsl The flag shouldn't be necessary for development AFAIK, only for running the code.
Is there anyone who's able to give me an example that I can run in Chrome 78, tried multiple things in a codesandbox environment on Chrome 78 but I can't declare an instance of the NFCReader.

beaufortfrancois, 2019-12-12:
I'll close as Chrome 79 is rolling out.
Please re-open if you still have some issues

closed 2019-12-12 by beaufortfrancois

richardrobberse, 2020-01-29:
@beaufortfrancois it seems the new API doesn't work in Chrome 79 either, can you please provide some more details?

beaufortfrancois, 2020-02-03:
"What web can do today" demos uses new `NDEFReader` and `NDEFWriter` interfaces which are only supported in Chrome 80 and onwards. See https://chromiumdash.appspot.com/commit/cde28a21cdcc8118b239c6de48b91a01fffe5e8f