w3c-cg/web-nfc #362: Change `toRecords()` to `records`

beaufortfrancois, 2019-10-03:
From a web developer perspective the `toText()`, `toArrayBuffer()` and `toJSON()` NDEFRecord methods seem to convert the record to a specified type while the recently added `toRecords()` method seems to be simple getter for a finite and non changing list of records. I understand it is technically not correct, this seems wrong to think that the NDEF Record is converted to a sequence of records.

For this reason, I think we should make it a readonly attribute and simply rename it to `records`. We could also use `children` but it's not my preferred choice.

I'm happy to send a PR if you agree.

```diff
  [Exposed=(Window)]
  interface NDEFRecord {
    constructor(NDEFRecordInit recordInit);

    readonly attribute NDEFRecordType recordType;
    readonly attribute USVString mediaType;
    readonly attribute USVString id;
+   readonly attribute NDEFRecords records;

    USVString? toText();
    [NewObject] ArrayBuffer? toArrayBuffer();
    [NewObject] any toJSON();
-   sequence<NDEFRecord> toRecords();
  };

+ [Exposed=(Window)]
+ interface NDEFRecords {
+   iterable<NDEFRecord>;
+   readonly attribute unsigned long length;
+   getter NDEFRecord (unsigned long index);
+ };
```

kenchris, 2019-10-03:
How is this different that a sequence of record? @marcoscaceres

beaufortfrancois, 2019-10-03:
I'm happy with this as well:

```diff
  [Exposed=(Window)]
  interface NDEFRecord {
    constructor(NDEFRecordInit recordInit);

    readonly attribute NDEFRecordType recordType;
    readonly attribute USVString mediaType;
    readonly attribute USVString id;
+   readonly attribute FrozenArray<NDEFRecord> records;

    USVString? toText();
    [NewObject] ArrayBuffer? toArrayBuffer();
    [NewObject] any toJSON();
-   sequence<NDEFRecord> toRecords();
  };
```

kenchris, 2019-10-03:
In a way it is converting (parsing) the records from the data... You can also get the raw data by calling toArrayBuffer().

Also I don't like the records attr to have side effects like parsing and caching first time, and doing it always if people won't use it, maybe doesn't make sense either

marcoscaceres, 2019-10-03:
About readonly FrozenArray:
https://github.com/heycam/webidl/issues/810

I'm also unsure why not just use a sequence (Array)? Alternative would be "takeRecords()", if you want to only allow taking the records once.

zolkis, 2019-10-03:
`toRecords()` will only work on some of the record types (TNF=1 "Sp" and TNF=4 or 5).
On the other record types it should fail.
I don't know if a property is a good way for representing that.

In that sense I like `toRecords()` since it means "I will try to parse the data payload of this record as it would be an NDEF message". That can either succeed (in a few cases) or fail (in most cases).

beaufortfrancois, 2019-10-03:
It seems as simple as checking the length attribute of `records`.

```js
if (record.records.length === 0) {
  // No children records
}
```

`toText()` doesn't fail either, it simply return null.
How is `toRecords` failing differently than returning an empty sequence?

zolkis, 2019-10-03:
It's fine to return `null`.
However, length check requires that the implementation checks whether the record could be parsed as a message before the user actually attempts that. 
For `"Sp"` that is fine since its a well known type. 
However, for external and unknown it is not fine. The best would be to have a method that can be invoked when the app knows this should succeed (since they have written the other end, for instance).

marcoscaceres, 2019-10-04:
> It's fine to return null.

Agree... but it's nice to just return an empty array. Less type checking, so if at all possible, just return an empty array.

zolkis, 2019-10-04:
Right, empty array is less type checking - but then how to differentiate an error condition from empty payload. OK, for smart poster the latter should be an error too (smart poster cannot be empty), but again, for TNF = {4,5} it could be the case (though making little to no sense).

However, my question is why not to use a method, once we need to run an algorithm to parse the data into records (that might also fail). It doesn't seem like a good match for representing it with a property (array).

zolkis, 2019-10-04:
Moreover, it doesn't work for majority of records.

If we go and represent with property, then we'd need to sub-class NDEFRecord and some of the sub-classes will have `record` and others won't. But even when they do have it, it won't work for every case of a given sub-class.

zolkis, 2019-10-04:
One more point (sorry for the flood).

As NDEF records are stored flat in an NDEFMessage and the payload length field is used to determine where a given record/message ends, it's a tree and not an array. Implementations need to parse the entire message to determine that record  tree.

In the current API we allow only smart records to be parsed into records in [9.4. Data Mapping](https://w3c.github.io/web-nfc/#data-mapping). 
Actually that could have a `records` property that should be non-empty.

But then how should `records` work with other record types? 
When a record doesn't have NDEFMessage as payload, then it's trivial, however, a bit messing with the `toText()`, `toJSON()` and `toArrayBuffer()` helpers.

The problem (again) is with external and unknown records. They will also have a `records` property, but it's not at all sure (actually is unlikely) they will have an NDEFMessage as payload. Now what should implementations do? Attempt parsing the payload as a message, just in case, that will fail most of the time?

To me it seems way simpler to just have a `toRecords()` method that may fail.
For the record, `toRecords()` is not my baby, IIRC it's Kenneth's FWIW, I just see it as good design.

marcoscaceres, 2019-10-04:
(again, I'm not looking at the details... I don't know anything about nfc so probably shouldn't comment  at all so apologies in advance and feel free to ignore me!) ... if there are errors, could an exception be thrown?

beaufortfrancois, 2019-10-08:
You convinced me that having an attribute was not the right choice for this.
Thank you @zolkis and @kenchris for your insights.

closed 2019-10-08 by beaufortfrancois