w3c-cg/web-nfc #363: Support Signature records

zolkis, 2019-10-03:
Related to #313

kenchris, 2019-11-06:
https://nfc-forum.org/purpose-nfc-ndef-signature-records/

https://en.wikipedia.org/wiki/Signature_Record_Type_Definition

zolkis, 2019-11-06:
The trust model Web NFC used to have so far is going to change.
Web NFC message and Web NFC id are already gone, instead signature records are coming in.
Security section will be updated soon.

Sig-records will work with app-level policy enforcement, since they are based on private keys. 
It will affect how the NFC content is handled at app level (including automated use cases).
(I don't know how many organizations will take the burden of signing, but it does offer content security.)

However, AFAICT they won't affect platform level automatic dispatching.

Since at the moment looks like this feature will not cause changing implementation-level security policies, just exposing signature records should be doable in this iteration. Or should it be a next-iteration feature?
 @leonhsl @riju

kenchris, 2019-11-06:
I think it would probably be good to do and can help enterprise adoption. As you said, most work is on the application side, we just need to spec and expose the record type

zolkis, 2019-11-06:
It would be also possible to do implementation-level policies, for instance if the message is signed, browser origin policy could be applied (similarly like we experimented with Web NFC id). This needs some more scrutiny, likely will not happen in this iteration, if at all.

zolkis, 2019-11-06:
We also need to support signing I guess. There are quite many rules in the Sig RTD. 
Do we allow sites to create invalid signed messages? 
Or do we spec algorithms for properly signing records? 
Or a helper API/flag to insert sig record? Based on site origin (https)?

Anyway at least we need an algorithm for creating (and validating?) Sig records.
I wonder could we improve/reuse the [Mapping local type to NDEF](https://w3c.github.io/web-nfc/#mapping-local-type-to-ndef).

kenchris, 2019-11-07:
I read that there has been some security issue with the first version of sig records, but I don't understand all details.

I do understand that by changing the payload size, you can make a trailing sig record move between being the last record inside am embedded record collection, or just  trailing the embedded record.

For instance
```
|[url]                               |[url]
|[smart poster]               |[smart poster with new payload length]
|  [t record]           --->    |  [t record]
|[sig record]                   |  [sig record]
```

zolkis, 2019-11-07:
But the sig record will still be for the whole message, including the url record and the smart poster.
IoW, after changing the smart poster payload length, the signature check will fail. As expected.

kenchris, 2019-11-07:
Yeah, that is what I was wondering how anyone would use that

zolkis, 2019-11-20:
The plan is to support creating and reading NDEF Signature records, but
- Web NFC won't sign any records or messages: applications need to do that
- Web NFC won't check any signatures: applications need to do that
- Key management is out of scope: applications need to do that.

The `data` property will access the raw bytes of the NDEF signature.
However, we could define the structure of a signature record (according to the Signature RTD spec) and include an algorithm to parse signatures (exposed as a new function on NDEFRecord, which kind of sucks). Also, creation of signatures could use the dictionaries defined for signature.

```javascript
dictionary NDEFSignature {
  required octet version;
  required Signature signature;
  required CertificateChain certificates; 
};

dictionary Signature {
  required boolean linked;  // true: data = URL, false: data = signature
  required octet type; // sig type mapped to integers 0..4
  required unsigned short length;  // 16 bits
  required ArrayBuffer data;  // signature or URL to the signature
};

dictionary CertificateChain {
  required boolean linked; // URI_Present in the Signature RTD
  required CertificateFormat format;
  USVString url;  // only present if linked is true
  required octet length;  // number of certs
  required sequence<ArrayBuffer> store;  // certificates: length, value
};

```

In the Signature RTD there are quite complex rules, too. 
We could deem this too complex and leave NDEF signatures completely in the applications' hands, including the structure of the payload (i.e. expose only the raw bytes and creating only from raw bytes - which sounds dangerous  without doing any validation).

Thoughts welcome.

beaufortfrancois, 2019-11-21:
Thanks for investigating @zolkis!
Could you provide a JS example of how reading and writing sig records would look like?

zolkis, 2019-11-21:
> Could you provide a JS example of how reading and writing sig records would look like?

That depends on what crypto library is used by the app. Since we don't know which one will be used, we cannot make a generic API for that, so we need to treat signatures as hex-strings. 
Those are needed to construct the (yet unspecified) `NDEFSignatureRecord`. 

Creating a signature requires a private key and a hex-string created from all participating record fields to be signed. Based on the security section, Web NFC should only support signing full NDEFMessage, instead of a list records, but the API would take a list of records. 
The signature record will be part of the NDEFMessage and constructed from the NDEFSignature dictionary filled up by the app. We could just have an extra optional `NDEFSignature` argument for `push()` that is prepared by the app?

On the receive side, the signature records will be like any other well known records. The verification process would fall on the app and would be quite tedious, as the app would need to fetch all the signable fields from all the records, then verify the signature against the (concatenated) data and certificate. If we want to make that more developer-friendly, we could expose a utility function to provide a hex-string of the data to be signed.

I made a few variants using the WebIDL [above](#issuecomment-556083175) and a [crypto library](https://kjur.github.io/jsrsasign/), but they are way too long (to be included in the spec) and also quite speculative at the moment as it would have little to do with the spec and more with the application. We might want to agree on some constraints (goals, means) before drafting examples.

kenchris, 2020-01-06:
People can write libraries around this if we just add a "signature" recordType and allow reading/writing raw bytes. It is the bare minimum.

If good solutions come from such libraries or we get extra cycles, then we can always add a toSignature() method and an API to easily deal with this