w3c-cg/web-nfc #367: Handling non-formatted tags

zolkis, 2019-10-10:
See also this [discussion](https://chromium-review.googlesource.com/c/chromium/src/+/1852215/1/services/device/nfc/android/java/src/org/chromium/device/nfc/NdefMessageValidator.java#47).

> Originally Web NFC was not meant to be a low-level NFC API and currently we don't explicitly handle unformatted/formatting use cases. 
I agree that presenting a non-formatted tag as an empty NDEF tag may be confusing if someone expects low-level NFC behavior from this API (there was a shift from high-level to lower-level recently, but not consistent). 
In that case we could take away that Note (from section 10.12.10, https://w3c.github.io/web-nfc/#the-nfc-reading-algorithm).

If we remove that Note, then a non-formatted tag will not produce any effect, i.e. no content should be presented, and no empty records should be presented (since there is nothing to read). 

However, whether writes are expected to succeed on non-formatted but NDEF-formatable tags, is a policy and the Web NFC API does not require enforcing that policy, in order that implementations could adapt to the underlying platform policies about this. Formally, implementations have freedom to decide about that, since they can abort a write for whatever reason in step 11.1 of the push algorithm: https://w3c.github.io/web-nfc/#the-push-method. Non-formatted tag MAY be one of them.

zolkis, 2019-10-10:
Options:
1. non-formatted tags are read as empty records (current behavior)
2. no action when reading non-formatted tag (since there is nothing to read).
3. writes to non-formatted tags will attempt formatting + writing (return error if fails).
4. writes to non-formatted tags will fail (then formatting is out of scope for the spec, you need to format it by external means).

I suggest 2+3.

reillyeon, 2019-10-10:
The problem with 2+3 as I understand from [Chromium issue 1012463](https://crbug.com/1012463) is that it doesn't allow an application to detect that there is data already on a card and will unconditionally overwrite it.

zolkis, 2019-10-10:
Non-formatted tags have no data on them.

Or do you mean that the problem is with 3, that it detects a tags as not formatted, but that actually is a different tech and then it will try to format it as NDEF and write NDEF on it? 

Implementations should be able to detect if a tag is really non-formatted. If already  implementations are still free to stop the push with an error.

There could be a number of policies on how to handle this, please suggest one if you have something in mind.

kenchris, 2019-10-10:
Sure? Isn't it not just that they are not NDEF formatted and have no NDEF data... Can't they have data using NFC-A,B,F,V etc ?

zolkis, 2019-10-10:
Right there was a [proposal](https://bugs.chromium.org/p/chromium/issues/detail?id=1012463#c8) for this:

> I was expecting a reading event with no data records in it to indicate an unformatted card (vs a single empty record for formatted but empty card).

So we could add 
5. when detecting non-formatted tag, fire an [NFCReadinEvent](https://w3c.github.io/web-nfc/#dom-nfcreadingevent) whose `message` contains an empty array for `records`.

Is that correct?

What should be the policy for pushing to a non-formatted tag?
We don't have an API for forrmatting a tag, so could we assume 3. (with eventual constraints - please suggest)?

zolkis, 2019-10-10:
> Isn't it not just that they are not NDEF formatted and have no NDEF data... Can't they have data using NFC-A,B,F,V etc ?

The point was that implementations can detect that condition and fail the write or add NDEF part (they can tell if a tag is NDEF-formattable).

So what we need to decide about is what policy do we choose.

reillyeon, 2019-10-10:
A major difference between the Android NFC API and the Web NFC API is that there is not a separate "card detected" event. This means that the _only_ way that a site can detect that a card is present is by attempting to read or write to it.

It is reasonable for a site to not want to unilaterally change the contents of every card it comes in contact with. The problem with the current policy is that it doesn't allow a site to detect that an unformatted card has been presented and so it _has_ to start with a write operation in order to be able to handle any card, or present the user with a "Did nothing happen? Press this button to format the card first." option.

Potential solutions include,
1. Specifying an event that is fired when an unformatted is presented while scanning.
2. Adding an option to push() that allows a site to detect when there is already data on a card (in any format?) and prevent overwriting it.

taralx, 2019-10-10:
As I noted in the Chromium bug, I expected a reading event with no records when an unformatted/unrecognized tag was presented. In other words, "we found a tag, here's its UID etc., but we didn't find any NDEF data on it".

As a user I would like to be able to distinguish "card cannot be formatted/written to" (possibly due to e.g. MIFARE keys being set), "card has data but it's not NDEF", and "card is unformatted (no data) but can be formatted".  Each of these may necessitate a different behavior in the application.

taralx, 2019-10-10:
I think it's fine if you want to keep the scope of the API to NDEF only. But there should be some kind of informative guidance on what a UA should do with tags that are out of scope. Perhaps the UA should tell the user "hey this tag isn't NDEF so I'm not telling the page about it"?

I would like a formatting API, but I understand the desire to avoid scope creep.

anssiko, 2019-10-10:
https://w3c.github.io/web-nfc/charter/#out-of-scope and https://w3c.github.io/web-nfc/charter/#charter-change is what we’ve agreed to. There’s some leeway in what constitutes low level.

Signaling of error conditions with a reason would be in scope of the current API given good use cases.

kenchris, 2019-10-10:
NDEF only for v1, but we want to be able to support more after shipping that, so we should keep that in mind

taralx, 2019-10-10:
Here's a couple of use cases:

1. Access control system based on card UID. Web console lets admins provision new cards into the system -- card contents are irrelevant, only need the UID on read.
2. "Flash drive"-style transfer app. Upload file, tap card to record URL. Need to be able to format blank cards as well as avoid accidentally overwriting data in the event the user taps the wrong card.

zolkis, 2019-10-10:
Charter change is possible, it's at least 30 days to process. The way I read [the Out of Scope](https://w3c.github.io/web-nfc/charter/#out-of-scope) section, we need to do that charter change for v2. It's a challenging task to come up with a new, broader scope that would be web-like, simple, with real, validated use cases, implementable, secure, etc., and get consensus on that.
Maybe we should set up some brainstorming about that as soon as this release is out.

Until then, we can update the spec if there is consensus or a decision about the desired behavior and policies.

I don't like the current Note about how non-formatted tags should be handled, either.

So we can do what @taralx proposed: on non-formatted tag read, fire an event with a message that has empty record array.
In addition, we can include an `overwrite` flag to push options  as @reillyeon suggested (which currently is `true`, and default value may remain the same, but apps would be able to set it to `false`).

We should also include notes / algorithmic steps that handle recent questions:
- what happens if other tech is on the tag that exposes NDEF (informative)
- when `mediaType` needs to be specified and when can it be empty (normative)
- FQDN in external type records (normative)
- will check if there is anything else.

Is that an acceptable solution for now @reillyeon ?

reillyeon, 2019-10-10:
I think it's the right path forward. We should have some kind of table explaining the various cases of unformatted, non-NDEF and NDEF cards being presented to read or write (with or without overwrite) just so I can be sure we're all on the same page.

Thank you @taralx for highlighting this issue.

zolkis, 2019-10-10:
A quick attempt out of my head (will double-check tomorrow, now go to sleep :), but you can comment ).
1. Reading tags with other tech and no NDEF exposed --> no events (ATM)
2. Reading tags with other tech + NDEF exposed --> event with NDEFMessage
3. Writing tags with other tech + no NDEF exposed --> error 
4. Writing tags with other tech + NDEF exposed --> error (should write via that tech)
5. Reading non-formatted tags --> event with empty records array for NDEFMessage
6. Writing non-formatted tags that are NDEF-formatable --> format + write, regardless of `overwrite` value. (*) 
7. Writing non-formatted tags that are not NDEF-formatable --> error
7. Writing NDEF tags with `overwrite = false` --> error

(*) Keep in mind this might fail as it's pretty experimental, i.e. OS and chip and tag dependent, since the NFC Forum does not specify this. Therefore we can't make any normative text on this, but the above should be ok on Android. So tags with other tech will not be NDEF-formatable AFAIK, but even if so, that would mean that formatting/writing would not remove other tech, so in all cases it can be assumed to fall in the writing NDEF category with no data to be overwritten, i.e. the `overwrite` flag has no relevance.
If anyone knows better, please correct me here.

taralx, 2019-10-12:
Seems reasonable. Anyone know what the iOS story looks like?

commit 7b659c8f (2019-10-16): Fix #367 (partial). Correct examples.
commit b71a81e3 (2019-10-16): Fix #367: add the 'overwrite' flag to NFCPushOptions and update the algorithms
closed 2019-10-16 by zolkis