w3c-cg/web-nfc #368: Notify user activation on NFC tap.

beaufortfrancois, 2019-10-11:
It would be nice if websites could react in an audible way when users tap NFC tags. With recent autoplay policy changes in browsers ([chrome](https://developers.google.com/web/updates/2017/09/autoplay-policy-changes), [safari](https://webkit.org/blog/7734/auto-play-policy-changes-for-macos/), [firefox](https://hacks.mozilla.org/2019/02/firefox-66-to-block-automatically-playing-audible-video-and-audio/)), I wonder if we should make NFC tap count as a user activation.

See https://mustaqahmed.github.io/user-activation-v2/

kenchris, 2019-10-11:
Like reconnect to pre-selected bluetooth device?

beaufortfrancois, 2019-10-11:
IIRC `device.gatt.connect()` doesn't require user gesture for now, but we should definitely require one if we're aiming to make it possible.

beaufortfrancois, 2019-10-11:
> A page can already make a sound when the push Promise resolves and there is an event for reads.

It's not true. If user didn't interact with the page first, sound is not allowed. See links above.

zolkis, 2019-10-11:
OK, sorry, so now the pages are gated unless user activation is sensed. 
Once "we make NFC tap count as a user activation" (by updating the implementation), then the gate is lifted and pages can play sound by reacting to Promise and by watching the event? (from the pov that does the API and spec need to change?)

Is there some standard wording that needs to be included for activation?) I assume this could be handled as a policy that is tied to a particular implementation, Android + Chrome in this case, and therefore could be encapsulated in the implementation, just like other UX related policies, so it should not creep up into the spec. Is that correct, or is there a reason for a spec change? (since the issue is filed against the spec).

beaufortfrancois, 2019-10-14:
@mustaqahmed may help us on the wording as I'm not sure either how to spec this.
I wonder if we should "simply" add a paragraph that extends https://html.spec.whatwg.org/multipage/interaction.html#triggered-by-user-activation

mustaqahmed, 2019-10-15:
That section of the HTML spec is [changing completely](https://github.com/whatwg/html/pull/3851), hopefully we will land it soon.  So I would suggest holding back new activation notifications now.

I have one question though: a user activation trigger is a (trusted) signal that the user has interacted with _a page_.  The NFC tap itself is not an interaction with the page but I believe it always follows an in-page interaction, right?  Then why should the tap also activate the page?  What am I missing here?

beaufortfrancois, 2019-10-16:
> That section of the HTML spec is [changing completely](https://github.com/whatwg/html/pull/3851), hopefully we will land it soon. So I would suggest holding back new activation notifications now.

Sure.

> I have one question though: a user activation trigger is a (trusted) signal that the user has interacted with _a page_. The NFC tap itself is not an interaction with the page but I believe it always [...]

When user land on a page that basically says "ready to scan", user move consciously their device towards the NFC tag to tap it, and this, for me at leasts, should count as an interaction with the page as user reacts to what the page says.

If I'm wrong, how could we make sure page can play an audible sound without user interaction when user tap NFC tag?

beaufortfrancois, 2019-11-05:
With https://github.com/w3c/web-nfc/issues/425 addressed, we can now close this issue.

closed 2019-11-05 by beaufortfrancois