w3c-cg/web-nfc #375: local types need some prefix of a kind

kenchris, 2019-10-15:
[image]

Look at the above. I can easily make a local type called "text" and then it will conflict with existing "text". NDEF binary gets around this by using uppercase for standardized well-known types and initial lowercase (or number) for local types, but that is very un-web-like.

A simple fix would be just add "local:" in front which kind of fits with external types "android.com:" etc

zolkis, 2019-10-15:
Local types have the local context (e.g. that of a parent record). 
Any "prefix" is application decision (e.g. to avoid conflicts).

Actually the NFC spec just talks about apps can "reuse" the local types (already defined in well-known types). It doesn't regulate anything else, except that they start with lowercase or number.

IMHO Web NFC should not impose any further restrictions here and we should not prefix local types.

kenchris, 2019-10-15:
We don't!

We use the name "text" for "T" as the web likes descriptive names/enums that are lowercase.

You can very well use a T well-known inside an embedded record, which we write using "recordType: text". Now what if I want to create a well-known type "text" - I cannot write it as "text" because that would become "T" and not "text" - this is just for us to describe that we want a local type.

zolkis, 2019-10-15:
It's the apps that should care about prefixing, not the spec. 

Not much related (since apps decide), but the example seems a bit extreme, why would apps want to define a local type "text" instead of reusing a text record? They just want confusion and more work?

beaufortfrancois, 2019-12-16:
@kenchris @zolkis Shall we add a note saying that Web NFC does NOT support local type names such as "empty", "text", "url", "smart-poster", "mime", "absolute-url", "unknown" to make it clear there's an actual restriction?

beaufortfrancois, 2019-12-16:
For info, the spec already says "a local type name value MUST NOT be equal to any other record types defined in this API."

[image]

kenchris, 2019-12-16:
This is the reason why I originally suggested that local names could start with : - like external names, they are kind of prefixed.

We might still read records with local types names equal to say "text" - who knows what exists out there

zolkis, 2019-12-16:
I think we don't need artificial constraints here, and starting exactly with `:` would overlap badly with external records (not long ago there was an idea that the implementation could generate the domain part and apps only specify `:<ext-type-name>`. 

The algorithms will first check on the record types in our spec, so there will be a match (causing an error later, potentially).
Then, if no standard names match, then we check for a valid external type name. If that is not a match, we check for a valid local name. At that point we won't have any overlaps, but we can still include in the validation steps. 
So I see no reason to introduce a new constraint here.

beaufortfrancois, 2019-12-16:
If we don't introduce a new constraint, shall we move the "a local type name value MUST NOT be equal to any other record types defined in this API." in the ["validating local type" algorithm](https://w3c.github.io/web-nfc/#validating-local-type) then?

zolkis, 2019-12-16:
That would definitely be my preference, too.

zolkis, 2019-12-16:
Actually the constraint in NDEF specs is that a local type name should not conflict in the given context. That doesn't necessarily mean it cannot be any of the standard names.
But it's good to make that statement, hopefully helping avoid very confusing situations.

beaufortfrancois, 2019-12-18:
@zolkis If a NDEF Record with local type "text" were to be discovered by the platform, how would it be translated for web developers in Web NFC?

```js
const record = externalRecord.toRecords()[0];
if (record.recordType == "text") {
    // Is it a TEXT record (TNF Field 1) or a local type "text" record?
}
```

zolkis, 2019-12-18:
If there is an external type TNF=4 record that contains other records, and one sub-record has record type `"text"` with TNF=1, then it is a WKT text record. It cannot be local type since it's already a text record.
One could not possibly write `"text"` with TNF=1 as local type either, since it's the definition of a text record.

In the example, one could use non-conflicting names for local type, e.g. write `"local:text"` or `"mytext"`.

beaufortfrancois, 2019-12-18:
> One could not possibly write "text" with TNF=1 as local type either, since it's the definition of a text record.

I thought it was possible to write "text" with TNF=1 as local type as a text record is "T", not "text".

zolkis, 2019-12-18:
Now since actually it's not `"text"` but `T` that is stored in the record, `"text"` could be a valid local type on NFC level. The constraint is added by Web NFC. 
We could possibly use `"T"` instead of `"text"` to denote text records in the Web NFC API, then someone would be able to use `"text"` as local type.

If we introduce any prefix for local types, it will conflict with another possible local type.

When we read, the implementation would know it's a `T` record, or is it local type with "text" and we could introduce a flag for marking it's a local type.

zolkis, 2019-12-18:
(cross-posting with @beaufortfrancois it seems)

beaufortfrancois, 2019-12-18:
Does that mean existing NFC tags with "text" local type records (not TEXT records) in an external type record will be wrongly recognized?

zolkis, 2019-12-18:
Yes. The implementation would know it, but currently it has no way to tell that to the app.
So we need either a flag or use `"T"` for text instead of `"text"`.

beaufortfrancois, 2019-12-18:
I'm not a huge fan of this flag but that seems to be the only option we'd have if we want to make Web NFC fully compliant with existing tags out there that may have "text" local type records in external records.

```js
const record = externalRecord.toRecords()[0];
if (record.recordType == "text") {
  // Is it a TEXT record or a local type "text" record?
  // Let's check additional property `isLocalRecord`
  if (record.isLocalRecord) {
    // It's a local type "text" record
  } else {
    // It's a TEXT record, let's use record.lang,
  }
}
```

zolkis, 2019-12-18:
There is another way, just use the NFC Forum WKT definitions: `T`, `U`, `Sp`, etc.

zolkis, 2019-12-21:
We need to make a decision between

1. Use TNF and TYPE fields to identify NDEF records (low-level). Other NFC APIs use this (e.g. using both `tnf` and `type`) with this [mapping](https://w3c.github.io/web-nfc/#data-mapping).

2. Maintain the current mapping between TNF+TYPE to the [recordType](https://zolkis.github.io/web-nfc/#the-record-type-string) string, which is more webapi'sh, but creates a clash with possible local type namespace. That conflict could be resolved by:

    a) using a flag like `boolean local` to tell a record is of local type 

    b) prefixing algorithm for `recordType`, e.g. choose a prefix string like `"_"` or `"local:"` and add/remove it consistently. For instance, a local type written as `"text"` would be specified as `"_text"` or `"local:text"` in the API. A local type written as `"_text"` or `"local:text"` would be specified as `"__text"` or `"local:local:text"`, respectively.

So the problem is that we don't use the NFC Forum standardized type names in the API, but invented our own (which we thought were more convenient for web developers). However, the scope of the API has moved from high-level to low-level, so IMHO now it should be fine using the NFC Forum standardized type names.

My take: I am not convinced the arbitrary complexity introduced by Web NFC in 2b) is worth doing, for the sake of saving complexity of defining both (standardized, i.e. already known) TNF and TYPE by applications.

Therefore my vote goes either for 1) or 2a).

beaufortfrancois, 2019-12-23:
@zolkis Well summarized!
After giving it more thoughts, I would go with the new boolean `local` flag to tell if a record is of local type to keep some webiness to the API.

kenchris, 2019-12-23:
I agree :-) I had a long argument but vacation kept me from writing anything so far

beaufortfrancois, 2019-12-24:
Does this IDL diff look good to you?

```diff
  typedef (DOMString or BufferSource or NDEFMessageInit) NDEFRecordDataSource;

  [SecureContext, Exposed=Window]
  interface NDEFRecord {
    constructor(NDEFRecordInit recordInit);

    readonly attribute USVString recordType;
    readonly attribute USVString? mediaType;
    readonly attribute USVString? id;
+   readonly attribute boolean isLocal;
    readonly attribute DataView? data;

    readonly attribute USVString? encoding;
    readonly attribute USVString? lang;

    sequence<NDEFRecord> toRecords();
  };

  dictionary NDEFRecordInit {
    required USVString recordType;
    USVString mediaType;
    USVString id;
+   boolean isLocal = false;

    USVString encoding;
    USVString lang;

    NDEFRecordDataSource data;
  };
```

zolkis, 2019-12-24:
Yes, it's good for me.

beaufortfrancois, 2019-12-24:
Can we merge this now that https://github.com/w3c/web-nfc/pull/491 has been merged?

zolkis, 2019-12-24:
Yes, we can close. @kenchris your's the honor, thanks for raising the issue and persisting with it :).

closed 2019-12-24 by kenchris