w3c-cg/web-nfc #387: NDEFRecord mediaType should be non null for media records

beaufortfrancois, 2019-10-18:
I'd like to make sure mapping is symmetric as much as possible and one bit that seems off is `mediaType`.

NDEFRecordInit mediaType is unused for all records but media.
However NDEFRecord mediaType is set for "text", "url", "external type", "local type", and "unknown".

```js
const writer = new NFCWriter();
await writer.push({ records: [
  {
    recordType: 'media',
    mediaType: 'application/octet-stream',
    data: new ArrayBuffer()
  },
  {
    recordType: 'text',
    data: 'hello'
  }
]});

const reader = new NFCReader();
reader.scan();
reader.onreading = event => {
  const decoder = new TextDecoder();
  for (const record of event.message.records) {
   if (record.recordType === 'media') {
    console.assert(record.mediaType === 'application/octet-stream');
  } else if (record.recordType === 'text') {
    // I didn't provide one so it should be empty.
    // And if I had provided a valid mediaType but different from 'text/plain' it would be unexpected.
    console.assert(record.mediaType === '');
    // Same would apply to "url"
  }
};
```

kenchris, 2019-10-18:
I thought about this too.

Media types are allowed to be empty (should not be null). You can actually associated a mime type (you cannot change it) with a lot of records, like text is text/plain

So the question also becomes, are we setting default media types when not explicitly stated? You can say that we do whether it is explicitly saved or not (it is saved implicitly as here we are using the T record). For that to make sense we could throw when someone adds a media type different from text/plain when storing "text".

The other option we have it to simply ignore media type for anything but "media" thus make it nullable amd throw if people try setting it. throw TypeError('mediaType doesn't apply for records of type "text"')

kenchris, 2019-10-18:
Regarding this, what I find important is that we can push() the read data within a reading event, and basically end up with the same data on the tag as before the push.

This is important if people want to update tags.

beaufortfrancois, 2019-10-28:
> The other option we have it to simply ignore media type for anything but "media" thus make it nullable amd throw if people try setting it. throw TypeError('mediaType doesn't apply for records of type "text"')

I'd like that.

We should also make sure our test suite covers the fact that pushing read data ends up with the same data on the tag.

closed 2019-10-28 by zolkis

beaufortfrancois, 2019-10-28:
@kenchris Hopefully https://github.com/w3c/web-nfc/pull/415 and https://github.com/w3c/web-nfc/pull/419 capture your feedback. Let me know if not and we can adjust.