w3c-cg/web-nfc #391: Avoid creating custom errors

kenchris, 2019-10-18:
We are told in other specs to avoid creating custom error types especially if they add nothing special

[image]

kenchris, 2019-10-18:
@rakuco just fixed wake lock spec in this sense

rakuco, 2019-10-18:
For more context: my commit fixed w3c/wake-lock#238 after following https://w3ctag.github.io/design-principles/#state-and-subclassing, as our `WakeLockEvent` had a single `lock` member that also happened to be identical to `Event.target` anyway.

I'm not very familiar with the NFC spec though, I guess if you were to drop `NFCErrorEvent` you'd still need to find a way to put that `error` attribute somewhere.

kenchris, 2019-10-18:
@reillyeon might have ideas

reillyeon, 2019-10-18:
In the Generic Sensors API the [`SensorErrorEvent`](https://w3c.github.io/sensors/#the-sensor-error-event-interface) is used to indicate that an error occurred while the sensor was active. The intuition is that a sensor can be initialized successfully and begin producing readings but then encounter and error (such as disconnection) and there needs to be a mechanism to report that. As the sensor has already been started successfully the Promise returned by the `start()` method has already settled, so an additional mechanism is needed.

In this specification I don't see `NFCErrorEvent` being thrown in any circumstance other than in the steps to initialize a scan. It is reasonable in that case to remove the event and have `scan()` return a Promise that is rejected instead. If, however, we want to report later error conditions we will need to have an event, just like in Generic Sensors.

I wonder if this concept is reusable to the point that the TAG would recommend standardizing a `DOMExceptionEvent` or `ErrorEvent` that could be reused across specifications for this purpose.

beaufortfrancois, 2019-10-19:
I agree with @reillyeon.

The funny thing is that push already rejects a promise with an appropriate DOMException, not a NDEFPushErrorEvent.

```js
const reader = new NDEFReader();
try {
  await reader.scan();
  // Scanning is happening...
} catch(e) {
  // NDEF scanning could not start.
}
```

I'll be happy to take action and update spec accordingly.

beaufortfrancois, 2019-10-19:
@kenchris Please assign this issue to me.