w3c-cg/web-nfc #393: Find a new name for MIME media-type records

beaufortfrancois, 2019-10-18:
https://github.com/w3ctag/design-principles/issues/140#issuecomment-543690443 suggests we should not use "media" for MIME media-type NDEF records. So let's pick another one:

- [ ] "media-type"
- [ ] "mime"
- [ ] "mime-media"
- [ ] "iana-media"
- [ ] "mime-typed-data"
- [ ] "typed-data"
- [ ] "mime-data"
- [ ] "mime-type"
- [ ] "geezer"

This list is not exhaustive. Feel free to add your suggestions.

kenchris, 2019-10-18:
Android uses MIME media in documentation and in some of the enums

zolkis, 2019-10-20:
I think "`mime`" is short and closest to the NDEF terminology, so I vote on that.

beaufortfrancois, 2019-10-21:
I'd go with `"mime"` as well.

zolkis, 2019-10-21:
OK. Let's work with this name for a while and consider others in the meantime.

commit b3035d23 (2019-10-22): Fix #393: change 'media' to 'mime'
commit b594ad4b (2019-10-22): Merge pull request #399 from zolkis/gh-pages

Fix #393: change 'media' to 'mime'
closed 2019-10-22 by zolkis