w3c-cg/web-nfc #408: Missing some record types in the algo creating ndef messages

leonhsl, 2019-10-24:
https://w3c.github.io/web-nfc/#creating-ndef-message needs to be updated to handle all possible record types, at least "smart-poster" and "absolute-url" are missing now.

zolkis, 2019-10-24:
Also, we need to provide algorithms to match external type and local type.

zolkis, 2019-11-26:
Also, we need algo for creating signature records, if we support that (see #363 for challenges).
Then only handover records are missing, should we support creating (and reading) them, too (as data)?

zolkis, 2020-01-03:
Algorithms for matching external and local types are in place now. Absolute-URL is missing.
Also, we need to complete the "parse Sp" steps (and modify the "parse record from bytes" on the way).
Filing separate issues to track these.

closed 2020-01-03 by zolkis