w3c-cg/web-nfc #409: What's the definition of "a standard local type to smart poster or a handover record"?

leonhsl, 2019-10-24:
5.2 of https://w3c.github.io/web-nfc/#dfn-map-local-type-to-ndef:

> If record's recordType is a standard local type to smart poster or a handover record, then set ndefRecord's TNF field to 1 (well-known type record), otherwise set ndefRecord's TNF field to 4 (external type record).

`recordType` is just a text starting with a lower case character or a number, how could we check whether it's `a standard local type to smart poster or a handover record`? There is no definition for `a standard local type to smart poster or a handover record` in the spec.

zolkis, 2019-10-24:
The standard local types, and also handover types (which we don't use for now) are enumerated in [this section](https://w3c.github.io/web-nfc/#well-known-type-records-tnf-1).
Should we have explicit algorithm to match external/standard local/local types?

leonhsl, 2019-10-24:
> Examples: "T" for text, "U" for URL, "Sp" for smart poster, "Sig" for signature, "Hc" for handover carrier, "Hr" for handover request, "Hs" for handover select, etc.

Do you mean the above enumeration? They are NOT local type names.

leonhsl, 2019-10-24:
Sure we just treat it as an external type record if the type string matches the rule of external types, no matter it's in an embedded context or not.

The question is for the following step, how could we tell it should be converted to external type record (TNF=4) or well-known type record (TNF=1)?
> otherwise, if a string is a match for being any local type (basically just starting with lowercase) and the context exists (currently: it is defined inside a message that is payload to another record), then it is local type

Probably the only case for TNF=1 should just be "the s, t and act in smart-poster", and handover types (we don't use now)? All other cases go with TNF=4?
But, from https://developer.android.com/reference/android/nfc/NdefRecord.html#TNF_WELL_KNOWN, it looks like Android API does not include "the s, t and act in smart-poster" in the scope of TNF=1, i.e. no RTD_XXX definition for "the s, t and act in smart-poster".

kenchris, 2019-10-24:
Local types means that types that are restricted to a certain other type of record. As they (their names and behavior) are restricted they don't need any namespacing as their names/types can be used inside other records with totally other meaning and data payload

beaufortfrancois, 2019-10-24:
See https://github.com/w3c/web-nfc/pull/411

kenchris, 2019-10-30:
Can we close?

closed 2019-10-31 by leonhsl