w3c-cg/web-nfc #418: Remove NDEFPushOptions timeout

beaufortfrancois, 2019-10-28:
The NDEFPushOptions `timeout` seems like it could be replaced with a simple timeout and NDEFPushOptions `signal` as you can see below.

Maybe there's more to it, but if not, shall we simply remove it?

```js
const writer = new NDEFWriter();

writer.push('hello world', { timeout: 5000 });
.catch(e => {
  if (e.name === 'TimeoutError') {
    // Push operation timed out.
  }
});
```
and its equivalent
```js
const writer = new NDEFWriter();
const controller = new AbortController();

setTimeout(() => controller.abort(), 5000);

writer.push('hello world', { signal: controller.signal });
.catch(e => {
  if (e.name === 'AbortError') {
    // Push operation has been aborted.
  }
});
```

zolkis, 2019-10-28:
I agree.

kenchris, 2019-10-30:
I already have been suggestion that earlier, so yeah, lgtm

beaufortfrancois, 2019-10-31:
I'll update spec then.

PR #428 (2019-10-31): Remove NDEFPushOptions timeout
FIX #418

***
<a href="https://pr-preview.s3.amazonaws.com/beaufortfrancois/web-nfc/pull/428.html" title="Last updated on Oct 31, 2019, 8:39 AM UTC (0ac7cce)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/428/0e251ee...beaufortfrancois:0ac7cce.html" title="Last updated on Oct 31, 2019, 8:39 AM UTC (0ac7cce)">Diff</a>
closed 2019-10-31 by kenchris