w3c-cg/web-nfc #420: Spec disagrees with Security section

zolkis, 2019-10-28:
The latest spec changes go against some statements in the [Security](https://w3c.github.io/web-nfc/#security-and-privacy) section.
Needs a complete overhaul, especially [Security policies](https://w3c.github.io/web-nfc/#security-policies), especially [storing site URL in record id](https://w3c.github.io/web-nfc/#store-site-url-as-record-identifier-when-writing-data).

zolkis, 2019-10-28:
We could also move the Security section to the end and at that, move out IDL index and Acknowledgement from under the previous section (alignment mismatch, see Table of Contents).

beaufortfrancois, 2019-11-05:
@zolkis Can you tackle this?

zolkis, 2019-11-05:
Yes, it's under work, with security people.

zolkis, 2019-11-06:
Related: #363 
Also: #438

zolkis, 2019-11-23:
Fixed by #440

closed 2019-11-23 by zolkis