w3c-cg/web-nfc #424: Remove NDEFPushOptions ignoreRead

beaufortfrancois, 2019-10-29:
It seems like the NDEFPushOptions `ignoreRead` could easily be replaced with NOT listening to the `onreading` event as you can see below.

If that's the case, shall we simply remove it?

```js
const reader = new NDEFReader();
reader.scan();
reader.onreading = event => { /* Won't be called because ignoreRead is true */ });

const writer = new NDEFWriter();
await writer.push('hello world', { ignoreRead: true });
```
and its equivalent
```js
const reader = new NDEFReader();
reader.scan();

const writer = new NDEFWriter();
await writer.push('hello world');
reader.onreading = event => { /* Will be called after write happened if needed */ });
```

zolkis, 2019-10-29:
Makes sense to me.

zolkis, 2019-11-26:
Do we go ahead with removing `ignoreRead`?

beaufortfrancois, 2019-12-18:
@kenchris Which changes would be required for your shopping demo for instance to handle this? Would this be that hard?

I'm trying to understand whether `ignoreRead` is convenient thing like we used to have with `toJSON`, etc. or if that's something web developers can't work without (or at least will have trouble with).

kenchris, 2019-12-18:
I find it annoying if not there. It will force me to handle all NFC in one place so basically create a singleton service.

The cost of having it is quite small

beaufortfrancois, 2019-12-24:
Argh! I'm torn ;)

On one hand, like you're saying, the cost of having it is quite small.
On the other hand, I feel like web developers would know how to do deal with this quite easily.
Even in your dialog case, it seems OK:

```js
const dialog = document.querySelector('dialog');

const observer = new MutationObserver(mutations => {
  myNdefReader.onreading = (mutations[0].target.open) ? null : onReading;
});
observer.observe(dialog, { attributeFilter: [ 'open' ] });

kenchris, 2019-12-24:
You assume that people deal with reading and writing in the same code and same UI parts, but that is seldom the case... It might even be done by different teams and even libraries in top

zolkis, 2019-12-24:
The use case seems to be that a writer could disable all reads for a given push. That would make `ignoreRead` global. However, we have a conflict between step 7.1 of [push()](https://w3c.github.io/web-nfc/#steps-push) and the [listening algorithm](https://w3c.github.io/web-nfc/#listening-for-content) saying 

> If there are any NDEFReader instances in activated reader objects then the UA MUST listen to NDEF messages on all connected NFC adapters.

zolkis, 2019-12-24:
BTW while checking this I realized we should check all occurrences of "NFC adapter" through the spec and make sure the parallel operation vs error reporting (for one or more adapters) is consistent.

zolkis, 2019-12-27:
See #494 .

zolkis, 2019-12-27:
Can we close (with #494 being merged)?

kenchris, 2020-01-02:
Let's close this

closed 2020-01-02 by kenchris