w3c-cg/web-nfc #425: push and scan should be gated by user activation

beaufortfrancois, 2019-10-30:
It looks like we're going with permission prompt. Therefore we should update spec to make sure `NDEFReader.scan` and `NDEFWriter.push `are gated by user activation.

@zolkis I think https://github.com/w3c/web-nfc/compare/gh-pages...beaufortfrancois:user-activation is not enough.  There's already some permission and user prompts section at https://w3c.github.io/web-nfc/#permissions-and-user-prompts which may also need some love.
Could you take a look?

This should solve #368

zolkis, 2019-10-30:
So this is a more specific formulation of #368?
Since the [HTML user activation PR](https://github.com/whatwg/html/pull/3851/files) has not landed yet, there is uncertainty what prose to use.
Should we encapsulate this anyway under "obtaining permission"?

BTW the whole security section needs some love, see #420

beaufortfrancois, 2019-10-30:
I think @mustaqahmed would know.

mustaqahmed, 2019-10-31:
Yes, "should be gated by user activation" is the right solution, addresses the concerns I had with #368.

Re fixing this spec issue: if the plan is to fix this before the user activation PR lands (hopefully in a few weeks), please use "triggered by user activation" for now and cc me in a new issue to rephrase it.  Or you can wait.

IFAIK Chrome's permission dialog is gated by user activation, at least for the cases I checked.  So the chain "NFC gated by permissions gated by user activation" makes perfect sense to me.  I am assuming there is no "piece" of the NFC API that does _not_ need a permission yet should be gated by user activation.

As a side note, I see that the [Permissions API](https://w3c.github.io/permissions/) does not specify the dependence on user activation.  @jyasskin, does it have an issue to cover this?

jyasskin, 2019-10-31:
https://github.com/w3c/permissions/pull/107 has resistance from Firefox folks to the idea of requiring user activation. https://github.com/w3c/permissions/issues/194 is also related.

beaufortfrancois, 2019-11-04:
@mustaqahmed We went for "triggered by user activation" for now. I'll open a new issue that I'll assign to you to rephrase it when user activation PR lands.

closed 2019-11-04 by beaufortfrancois