w3c-cg/web-nfc #429: User feedback: PWA integration

kenchris, 2019-10-31:
When tapping a NFC tag which has URL record in order to launch the site/app, then it would be nice that a second tap isn't needed

not running PWA (launch):
> records were read and used before app called scan()

running PWA (should gain focus, if not focused):
> records were read but probably didn't cause onmessage event due to lack of focus or because the system handled the event

zolkis, 2019-10-31:
How can we connect the previous scan with the current use in a secure way?

kenchris, 2019-10-31:
I am not sure how to accomplish this on Android, but I definitely understand the use-case. This probably requires some special handling code in the Web APK parts

zolkis, 2019-10-31:
Let's do a time sequence:
1. user taps tag with URL record (first record)
2. system service handles intent, and opens browser with the URL. 
At this point Web NFC implementation is likely not running yet.
3. PWA comes to senses, Web NFC implementation, too, and want to be fed with the "parent" tag's data.

So there must be a separate service running that can identify and match the PWA with the "parent". That has to be a system service, maybe part of Web NFC implementation. Should Web NFC (in this mode) run as a service once a read is made?

beaufortfrancois, 2019-10-31:
Is this related to https://github.com/w3c/web-nfc/issues/78#issuecomment-546827632?

kenchris, 2019-10-31:
Yes that would be related

zolkis, 2019-10-31:
OK, so we have 2 problems:
- how to expose a tag read in the past but which is connected to this page by bootstrapping it
- how to implement this under the hood.

About exposing, I don't like the idea of an event that would represent a reading that happened before the event listener was instantiated. In addition, we are introducing a new interaction model with NDEF, using events: we should either go full monty on that - which might be a good idea -, or stick with the current interactions. 

IMHO it would be better to model this use case by providing the already available NDEF data e.g. using a `referrerNDEF` property suggested by @kenchris. Granted, that would only work with NDEF (whereas an event would work with any tech), but that's not an issue any more since we only have NDEFReader.

We need a PoC for the implementation, then we need to decide whether the benefit is worth the trouble (which I am not sure about, since some existing Android NFC apps do require 2 taps to work, so it's not an unknown pattern to NFC users; on the other hand, and I am not sure whether we can really do it without 2 taps in a clean way). I agree it would be nice.

kenchris, 2019-11-01:
The whole launching a url is NDEF specific, so I guess that is fine :)

kenchris, 2019-11-04:
From @zolkis:

> https://www.zdnet.com/article/android-bug-lets-hackers-plant-malware-via-nfc-beaming/
> The morale: convenience vs security is not childplay. Like making content available with opening URL. We need threat analysis for the use case.

I don't think this is a big risk here.

* You cannot install anything from NDEF via out API
* Data is inert, so if you want to instantiated/evaluate WASM/JS from a tag, then the website will have to consider the security implications and as such probably make sure the data is signed (a record exists for that)

zolkis, 2019-11-04:
What we need to ensure is the connection between the tag+URL and the NDEF data exposed to a site. For instance, in the following scenarion,
- page 1 uses Web NFC, has a reader set up
- a tag containing a bootstrap URL comes in range
- a new page 2 is launched by the browser from the tag, exposing the tag data to it
- but site 1 will also get the data since it has a reader set up.

Only page 2 should be "connected" to the tag, if that has any meaning related to "trust" (we have not defined that yet). There is _some_ assumption there, since we "save" another tap before reading tap data, therefore going around the security policies that are in place for tags in general.
At the moment we say tags integrity is not to be trusted, but with this use case, we need to clarify if the data and page have any trust relationship. What does it mean to have  a `referrerNDEF` or something like that. 

I'd leave threat assessment to security people. 

This is not an argument against the feature, just a note that threat assessment must be done for such new features.

zolkis, 2019-11-26:
Do we have a proposal how to move on with this?

kenchris, 2020-05-20:
Duplicate of #78

closed 2020-05-20 by kenchris