w3c-cg/web-nfc #434: What is the correct way to check whether the browser supports Web NFC API or not?

FluorescentHallucinogen, 2019-11-05:
What is the correct way to check whether the browser supports Web NFC API or not?

In the previous version of API, I've used the following:

```js
if ('nfc' in navigator) {
  // use Web NFC API
}
```

Is

```js
if ('NDEFReader' in window && 'NDEFWriter' in window) {
  // use Web NFC API
}
```

the correct way?

It would also really nice to have a property (or something) to check whether the device has NFC chip or not.

FluorescentHallucinogen, 2019-11-05:
Are there NFC chips that support only reading, but not writing NFC tags? If yes, how to detect this?

beaufortfrancois, 2019-11-05:
Indeed!
```js
if ('NDEFReader' in window) { /* ... Scan NDEF Tags */ }
if ('NDEFWriter' in window) { /* ... Write NDEF Tags */ }
```
You can also use Permissions API
```js
try {
  const status = await navigator.permissions.query({ name: 'nfc'});
}
catch(e) {
  // No Web NFC support
}
```

beaufortfrancois, 2019-11-05:
To check whether the device has NFC chip or not, you can call `scan()` or `push()`and it will reject with an appropriate error if that's the case.

beaufortfrancois, 2019-11-05:
> Are there NFC chips that support only reading, but not writing NFC tags? If yes, how to detect this?

I don't think so.

FluorescentHallucinogen, 2019-11-05:
@beaufortfrancois What about adding this to the spec?

beaufortfrancois, 2019-11-05:
Which parts?

Good luck for your [talk](https://twitter.com/GDGNsk/status/1187389660462485505) by the way ;)

FluorescentHallucinogen, 2019-11-05:
@beaufortfrancois 
> Good luck for your talk by the way ;)

Thanks! This is workshop, not talk. :)

> Which parts?

About browser feature detection: https://github.com/w3c/web-nfc/issues/434#issuecomment-549765910.

beaufortfrancois, 2019-11-05:
I'll do this.
Feel free to file more issues as you play with Web NFC.
Note that current Chromium implementation doesn't show prompts yet, but will really soon.

FluorescentHallucinogen, 2019-11-05:
> Note that current Chromium implementation doesn't show prompts yet, but will really soon.

I [know](https://chromium.googlesource.com/chromium/src.git/+/51f9ee3727aefb076947fe6ab58154f479bc30d3). :) Thanks!

beaufortfrancois, 2019-11-05:
Cool!
You can check WIP at https://chromium-review.googlesource.com/dashboard/beaufort.francois@gmail.com

FluorescentHallucinogen, 2019-11-05:
@beaufortfrancois Just curious, what if I call `navigator.permissions.query({ name: 'nfc'})`, but NFC is disabled in the OS settings? Does your CL cover this use case similar to the following (example for Web Geolocation API)?

[image]
[image]
[image]
[image]

beaufortfrancois, 2019-11-05:
`navigator.permissions.query({ name: 'nfc'})` will resolve with a status of "denied".

```js
navigator.permissions.query({ name: "nfc" }).then(permissionStatus => {
  console.log(`NFC user permission: ${permissionStatus.state}`);
  permissionStatus.onchange = _ => {
    console.log(`NFC user permission changed: ${permissionStatus.state}`);
  };
});
```

beaufortfrancois, 2019-11-05:
https://w3c.github.io/web-nfc/#feature-support has been added to the spec.

closed 2019-11-05 by beaufortfrancois

FluorescentHallucinogen, 2019-11-05:
@beaufortfrancois 
> `navigator.permissions.query({ name: 'nfc'})` will resolve with a status of "denied".

Silently? It would be great to let the user know what went wrong and provide the ability to enable NFC in the OS settings (see screenshots in https://github.com/w3c/web-nfc/issues/434#issuecomment-549795888).

P.S. Sorry for offtopic.

beaufortfrancois, 2019-11-05:
`navigator.permissions.query` will always be silent.  What we could use is `navigator.permissions.request`, however it is not functional at the moment in Chromium.

Having something similar to Geolocation makes sense to me.
I'll ping appropriate folks. Thanks!

FluorescentHallucinogen, 2019-11-21:
@beaufortfrancois 

Just to clarify, will there be a separate permission prompt for writing NFC (alongside permission prompt for reading NFC) or one single permission prompt for the whole Web NFC API (both reading and writing)?

Use case: what if I want to allow the web app to only read NFC tags but not write?

I'm not familiar with the process of merge code in Chromium. I see that the NFC permission change [was merged](https://chromium-review.googlesource.com/q/hashtag:%22webnfc%22), but I've just tested in Chrome Canary for Android with `chrome://flags/#enable-webnfc` flag enabled today and for some reason don't see the permission prompt. Why? Any ETA?

beaufortfrancois, 2019-11-21:
> Just to clarify, will there be a separate permission prompt for writing NFC (alongside permission prompt for reading NFC) or one single permission prompt for the whole Web NFC API (both reading and writing)?

In Chromium, there will be one single permission prompt for using the Web NFC API.

> I'm not familiar with the process of merge code in Chromium. I see that the NFC permission change [was merged](https://chromium-review.googlesource.com/q/hashtag:%22webnfc%22), but I've just tested [...]

I'm currently working on making sure tests pass for https://chromium-review.googlesource.com/c/chromium/src/+/1895345. Once it's solved, it will be pushed to Chromium. I hope this will be done this week or next week.

FluorescentHallucinogen, 2019-11-21:
@beaufortfrancois
> In Chromium, there will be one single permission prompt for using the Web NFC API.

Any plans to separate permission prompts for reading and writing?

From the name of https://chromium-review.googlesource.com/c/chromium/src/+/1895345 — `[WebNFC] Add permission prompt for NDEFWriter.push` it's not clear is this for writing only?

What if I try to read NFC tags (if user didn't previously allow Web NFC), will the permission prompt be shown?

beaufortfrancois, 2019-11-21:
> @beaufortfrancois
> 
> > In Chromium, there will be one single permission prompt for using the Web NFC API.
> 
> Any plans to separate permission prompts for reading and writing?

Nope. Not in Chromium.

> From the name of https://chromium-review.googlesource.com/c/chromium/src/+/1895345 — `[WebNFC] Add permission prompt for NDEFWriter.push` it's not clear is this for writing only?
> 
> What if I try to read NFC tags (if user didn't previously allow Web NFC), will the permission prompt be shown?

Yup. Once user approves Web NFC, prompt won't be shown anymore.

beaufortfrancois, 2019-11-29:
For info, I'm currently adding "second" prompt to Chrome to turn on NFC if needed. See https://chromium-review.googlesource.com/c/chromium/src/+/1940241

[image]

beaufortfrancois, 2019-12-18:
@FluorescentHallucinogen All bits have finally [landed](https://chromiumdash.appspot.com/commits?user=beaufort.francois%40gmail.com&platform=Android) in Chrome Canary for Android. You can give it a try at https://permission.site

[image] [image] [image] [image]

FluorescentHallucinogen, 2019-12-18:
@beaufortfrancois That's awesome! Thanks for your work!

One detail: what about changing the word "phone" in texts to more generic "device"? What if my device is e.g. tablet, not phone?

beaufortfrancois, 2019-12-18:
@FluorescentHallucinogen I'll share this feedback with the team. Thanks!

FluorescentHallucinogen, 2019-12-26:
@beaufortfrancois Any plans on Origin Trial for Web NFC API?

riju, 2019-12-26:
@FluorescentHallucinogen : We are planning for M81, so around end of Jan '20.

beaufortfrancois, 2020-03-05:
To know more about the Origin Trial for Web NFC, I'd recommend having a look at https://web.dev/nfc/