w3c-cg/web-nfc #437: scan() should validate NDEFScanOptions#{recordType,mediaType}

leonhsl, 2019-11-06:
For example, `mediaType` should always be `undefined` for record types other than `mime`, otherwise we just throw a TypeError like what we do in constructing an `NDEFRecord`.

beaufortfrancois, 2019-11-06:
I understand where this is coming from, but I'm not sure we should throw a TypeError for scan options as this is technically NOT an invalid filter. It's just that scan won't trigger any onreading event.

I'm afraid we'd make it more complicated to cover all cases.
For instance, should we throw a TypeError when NDEFScanOptions `recordType` is set to `FOO`? 

I'll let other chime as I don't feel strongly about it.

zolkis, 2019-11-06:
>  I'm not sure we should throw a TypeError for scan options as this is technically NOT an invalid filter. It's just that scan won't trigger any onreading event.

I agree with this. The scan algorithm is sufficiently tight IMHO.

leonhsl, 2019-11-06:
I also prefer simplicity.

Take this scan options `{recordType: 'url', mediaType: 'text/plain'}` as an example, we have 2 choices for matching 'url' records (whose media type is always null):
  1. ignore the careless user input (mediaType) and match all 'url' records.
  2. do strictly, i.e. do not match any 'url' records.
I'd prefer 1 because we are already aware that mediaType does not make any sense for 'url' records at all, if we do not notify users of the invalid input then we'd better not fail silently when matching.

Either way, the spec needs to make a rule/note explicitly in the section of NDEFScanOptions definition.

zolkis, 2019-11-06:
What happens now is that properties defined in `option` are passed to the internal slots. No checks here. I guess this is what you'd like to get improved. 

Then, when a tag is read, the [Dispathcing NFC content](https://w3c.github.io/web-nfc/#dispatching-nfc-content) is run, which will first test for `id` equality for _any_ records of the message, then for `recordType` equality over _any_ records of the message, then for `mediaType` equality over _any_ records of the message. All that are specified in the internal slots must be equal to the values from the tags. Only then an event is fired.

This means that if `options` specifies `Foo` for `mediaType`, then the tag will have to have a record with `mediaType` being `Foo`, otherwise it's not a match. 

Since we don't validate `mediaType`, there is not much difference validating against empty `mediaType`, since it will be automatically ignored.

I am not sure what would be the added value if validation was stricter and earlier.

zolkis, 2019-11-06:
Another point is that `NDEFScanOptions` is a property bag and does not need to be validated as `NDEFRecord`.

But a clarifying note is possible.

zolkis, 2019-11-26:
@leonhsl do you expect changes on this?
IMHO the algorithmic steps are sufficiently specified.

leonhsl, 2019-11-27:
https://github.com/w3c/web-nfc/issues/437#issuecomment-550222603 reminds me to read [Dispathcing NFC content](https://w3c.github.io/web-nfc/#dispatching-nfc-content) carefully and now I feel the algorithm is incorrect.

Consider we have a scan options `{id: 'dummy-id', recordType: 'url'}`, my understanding (intuitive feeling) is that a message matches only if it contains a `url` record with `dummy-id`, but the current algorithm suggests that a message containing a whatever (`text`, `unknown` etc.) type record with `dummy-id` will just match.

This is sort of a surprise to me, and I'd like to know opinions from all. @beaufortfrancois @kenchris @riju

beaufortfrancois, 2019-11-27:
Re-reading "Dispatching NFC content" algorithm below suggests that an NDEF message will matches only if it contains a url record and a record with dummy-id. It may be the same record, but not necessary.

> For each NDEFReader instance reader in the activated reader objects, run the following sub-steps:
> 
> 1. If reader.[[Id]] is present and it is not equal to any record's id where record is an element of message, continue.
> 2. If reader.[[RecordType]] is present and it is not equal to any record's recordType where record is an element of message, continue.
> 3. If reader.[[MediaType]] is not "" and it is not equal to any record's mediaType where record is an element of message, continue.
> 4. Fire an event named "reading" at reader using NDEFReadingEvent with its serialNumber attribute initialized to serialNumber and message attribute initialized to message.

zolkis, 2019-11-27:
> Re-reading "Dispatching NFC content" algorithm below suggests that an NDEF message will matches only if it contains a url record and a record with dummy-id.

I understand the same way.

If `[[Id]]` present, must match one of the records `id`, otherwise no match (event never fired), continue with the next reader.
If `[[RecordType]]` present, must match one of the record `recordType`, otherwise no match, continue with the next reader.
If `[[MediaType]]` is given, must match one of the record `mediaType`, otherwise no match, continue with the next reader.
(If all conditions above matched, fire an event).

BTW there is a bug in the algo, `[[Id]]` should be tested against empty string, not for presence.

leonhsl, 2019-11-27:
The discussing point now is whether the conditions (inside the scan options) should be fulfilled by the same record or they can be scattered across several records. I personally feel the latter option is weird.

zolkis, 2019-11-27:
It is the latter. 
We might include a reader option to specify the type of matching: all conditions per record, or all conditions per message (default, to match the current behaviour).

leonhsl, 2019-11-29:
Do we have other opinions here? I still think `id, recordType, mediaType` is a property set that's only meaningful as a whole, matching them separately on different records does not make sense to me and Web Devs would be confused about the behavior, but in any case I'd make the impl follow most people's opinions here.

FYI, the impl has always been option 1: the conditions should be fulfilled by the same record.

zolkis, 2019-11-29:
I am not against modifying the filtering so that if a filter has more than one property, a message is a match only if it contains at least one record that matches all of the filter properties. 

Other than that, should we have filter policy options (like to select on of the behaviors above)?
What are typical use cases of multi-property filters?
The current algo permits filters like "give me text records **or** media type with `application/json`.
Is that useful, when one could set 2 readers, one with text, one with media?
So Leon's suggestion makes sense to me.
@kenchris @beaufortfrancois ?

beaufortfrancois, 2019-12-02:
I'll vote for @leonhsl suggestion to make conditions fulfilled by the same NDEF record.

beaufortfrancois, 2019-12-13:
@kenchris (gentle ping)

kenchris, 2019-12-13:
I totally agree, they should be for the same record, otherwise there is not much sense in adding a filter.

Maybe it should make the filter more explicit and allow multiple?

```js
{ filter: [ { recordType: "mime"}, { recordType: "url", id: "example.com" } ] }
```

zolkis, 2019-12-13:
Specifying multiple Readers would end up being the same thing (OR'ing filters), and it's arguably a simpler design. It also simpler for clients, since they don't have to look which of the complex filter was satisfied. With simple filters per Reader that's a given.
So it depends on the expected usage patterns.

zolkis, 2019-12-19:
Just a note, #481 should fix the dispatching algorithm (side discussion here).
However, the scan options could be also validated in the [scan](https://w3c.github.io/web-nfc/#the-scan-method) algorithm (the main issue here) if we can come up with a relatively simple validation rule set.

zolkis, 2019-12-27:
@leonhsl can we close this now?

closed 2019-12-30 by leonhsl