w3c-cg/web-nfc #438: obtain reading and push permission should be the same

beaufortfrancois, 2019-11-06:
I'm not sure why but "[obtain reading permission](https://w3c.github.io/web-nfc/#dfn-obtain-reading-permission)" and "[obtain push permission](https://w3c.github.io/web-nfc/#dfn-obtain-push-permission)" are different. I think both should be the same and consolidated under a unique "obtain nfc permission".

What are your thoughts?

zolkis, 2019-11-06:
Isn't pushing more powerful feature than reading? 

I think originally the idea was that reading policy could be more relaxed than writing policy.

Especially with signature records, we could apply browser origin policies against the URLs contained in the cert chain and/or the URL record in the message. I guess read and write scenarios could undergo different policies?

riju, 2019-11-06:
my 2 cents: NFC read and write should be the same permission at least say Chrome's PoV, maybe we can add a note that implementer is free to have a choice but the first implementer Chrome is going to use same `nfc` permission

kenchris, 2019-11-07:
Permission allow to have options along with them no? If so, it can always be extended

zolkis, 2019-11-23:
Fixed by #440

zolkis, 2019-11-26:
Can we close this?

beaufortfrancois, 2019-11-26:
I think so.

closed 2019-11-26 by zolkis