w3c-cg/web-nfc #442: Can't read empty tags

joshdillner, 2019-11-15:
Hi, I've been implementing the latest API and testing on Chrome Canary, and I've found that I can't scan empty tags - it never enters the "onreading" or "onerror" event.

I've also found that writing a url record type seems to add 2 records:
Record 0, type = URI record
Record 1, type = w3.org:a

Can you advise? Here's my code:

(async () => {
  this.setState({error: {message: 'ready to scan'}});
  
  await reader.scan({ signal: controller.signal });
  
  reader.onreading = event => {
    this.setState({error: {message: 'found tag, starting write'}});

    const writer = new NDEFWriter();
      
    writer.push({
      records: [{
        recordType: 'url',
        data: `https://www.google.com`
      }]
    }).then(() => {
      this.setState({writing: false, writeResult: 'success', error: {message: 'write success'}});
      controller.abort();
    }).catch(e => {
      this.setState({writing: false, writeResult: 'fail', error: {...e, message: 'writing failed, try again'}});
      controller.abort();
    });
  }

  reader.onerror = event => {
    this.setState({error: {message: 'scanning error'}});
  }
})();

beaufortfrancois, 2019-11-18:
Is your empty tag NDEF formatted?
If not, this may explain why you don't receive any event.
If so, you should receive an "onreading" event with one "empty" message. See https://chromiumdash.appspot.com/commit/645a24e453cbb8767c7684527c48359b23ecac1f
Make sure you use latest Chrome Canary on Android.

Regarding the second record, this happen because previous version of the Web NFC spec included a "special" record when writing from Web NFC. This is no longer the case and will be removed from the Chromium implementation.

beaufortfrancois, 2019-11-26:
Please re-open if you're still having issues.

closed 2019-11-26 by beaufortfrancois