w3c-cg/web-nfc #445: Keep (how?) or remove the record id?

leonhsl, 2019-11-22:
This issue comes from the discussion at https://github.com/w3c/web-nfc/pull/444#discussion_r349416140

leonhsl, 2019-11-22:
As we've already been making WebNFC kind of a low-level API so it's good to abandon all those 'smartness' left from the old era, like binding record ids with the hostname of the current web page.

+1 to the following suggestion from @beaufortfrancois 

> I'd be in favor of removing this "id" constraint algorithm and let web developers do what ever they want with it.

const writer = new NDEFWriter();
await writer.push({ records: [{ recordType: 'text', data: 'hi', id: '123' }] });
...
const reader = new NDEFReader();
await reader.scan({id: '123' });
reader.onreading = event => {};

beaufortfrancois, 2019-11-23:
If we all agree on removing constraint on record id, I'll assign the task of updating the spec to myself.

beaufortfrancois, 2019-11-25:
I've started updating spec at https://github.com/w3c/web-nfc/pull/446

beaufortfrancois, 2019-11-25:
@leonhsl Let us know if that looks good to you.

PR #446 (2019-11-25): Remove URL constraint for record id
As discussed in https://github.com/w3c/web-nfc/pull/444#discussion_r349416140, this PR removes binding record ids with the hostname of the current web page.

FIX #445

***
<a href="https://pr-preview.s3.amazonaws.com/beaufortfrancois/web-nfc/pull/446.html" title="Last updated on Nov 25, 2019, 9:03 AM UTC (a78d5fb)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/446/f28bf10...beaufortfrancois:a78d5fb.html" title="Last updated on Nov 25, 2019, 9:03 AM UTC (a78d5fb)">Diff</a>
closed 2019-11-25 by beaufortfrancois